EdgeTX/edgetx #2108: Some issues to fix on Frsky Horus X12S

Luznatural, 2022-06-30:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

The following issues need fix in EdgeTX for Frsky Horus X12S Transmitters:
1. Issues previously working OK in all OTX versions, that are not working now on ETX:
- In the tab Hardware (Sys-Hardware) the option "Antenna" for choosing between the internal  or the external antennas is missing. The issue should appear inmediately after bluetooth, with the following selections: "Ask", "Per model", and "Internal + External"
-  When performing a calibration (Sys-Hardware-Calibration), you are unable to calibrate the third gimbal (the one located in the lower left part of the radio). This gimbal does not even appear at all in the screen.
- In "Model Setup" the "Internal RF tab" should read "iXJT" (acronysm for "Improved XJT"). This internal module of the X12S is much more modern than the XJT one, and has communications capability with the radio (for communicating the firmware version of the module, that also does not appear (as an example).
- In "Model Setup", inmediatelly after "receiver", the "Antenna" selection is also missing, with the following selections: "Ask", "Internal", and "Internal + External".
2. New problems detected:
- When pressing "Model Select", it takes too much time (far more than 10 seconds) for the different models to appear in the screen.
-  If you have more than one category of planes, it takes too much time to move from one to the next (far more than 10 seconds) when you press "Page Up" or "Page Down".
-  Same happens when you want to select "create a new category", as it takes so long that you frequently think that the radio is locked.

NOTE: I offer myself to test corrections as necessary. All occurrences appear in a stock Frsky Horus X12S without touch screen kit, nor ACCESS upgrades.

Thanks in advance

### Expected Behavior

I Expect everything working as in OTX versions

### Steps To Reproduce

As described above

### Version

2.7.1

### Transmitter

FrSky X12

pfeerick, 2022-07-01:
FYI, it's best to keep issues focused on one topic, otherwise they will invariable only be partly resolved when fixed if they cover disparately different topics. 

Regarding the slow model select behaviour, this is a known issue which is across the board with models, and gets progressivly worse the more models with images you add to a category (as it is the loading of images which is causing the slowdown). I haven't checked if there has been any change on this in the LVGL re-write, but if it hasn't, it is a known issue that needs to be resolved either through UI giving feedback or some form of image caching/buffering to speed up the loading. 

The missing antenna selection issues overlap with https://github.com/EdgeTX/edgetx/issues/1185 (for the X10), so should hopefully resolved once the associated PR is rebased with current state of development and can be properly tested.

Luznatural, 2022-07-01:
Thanks for taking time for responding. You are totally right: is far better to separate issues. I will do that in the future.

The ability to select antennas is of extreme importance, as every antenna has a separate power transistor that might burn if you do not have attached the physical antenna to the external plug. Normally, you fly without attaching the external antenna, unless you want to enhance emission distances. This issue was solved very early in OTX, well before ETX fork was created. Would it be possible to replicate that solution to ETX? I read #1185: I will try to use Companion to check if the external antenna is working, and as a default will always fly with the external antenna plugged.

Same happens with the ability to calibrate and use the third gimbal. Would it be possible to replicate the solution from OTX?

Knowing the origin of the problem for slow model selection, I think is something I can live with, until fixing. It,s a matter of making categories with lesser number of planes, and have more patience.

Thanks again for your response, offering help to test as neccessary.
Very Respectfully

gagarinlg, 2022-07-01:
> The ability to select antennas is of extreme importance, as every antenna has a separate power transistor that might burn if you do not have attached the physical antenna to the external plug. [...]

These options were missing in the OpenTX 2.4 dev branch and EdgeTX was created from said branch.

raphaelcoeffic, 2022-07-04:
> In "Model Setup" the "Internal RF tab" should read "iXJT" (acronysm for "Improved XJT"). This internal module of the X12S is much more modern than the XJT one, and has communications capability with [...]

There is no single change compared to a "normal" internal XJT from a protocol point of view. We're not here to implement FrSky marketing ;-) OpenTx used to have special treatment for FrSky, but this is over. FrSky does not cooperate with us at all. So please consider this as "community maintained", basically meaning "best effort" of whoever has time for this. If you want full support with the marketing acronyms, please use Ethos, this is what FrSky does actively support.

raphaelcoeffic, 2022-07-04:
> When pressing "Model Select", it takes too much time (far more than 10 seconds) for the different models to appear in the screen.

This is basically caused by the images being slow to load. We will add a progress bar in 2.8, and possibly generate thumbnails later on. Down-scaling the images does take some time as well.

brainbubblersbest, 2022-09-25:
It works with isrm. If you like the latest Stuff available, then maybe an Upgrade HF make sense. NOTHING is for free and FRSKY is very uncooperative. Im thankfull for every work on FRSKY specific issues and because of that, without a question, make a monthly donation. (Support doesnt mean you get anything to be clear)
I hope at least Bsongis like the fact, that his Baby gettin Better and older after it has done its first steps. Even the Community gettin growth a few Times bigger then before, like RM and Jumper Pilots. Surprise Surprise FRSKY.

raphaelcoeffic, 2022-09-25:
Worth noting regarding antenna selection: #1765

raphaelcoeffic, 2022-09-25:
All the other items regarding `Model select` / `Categories`: fixed with new `Model labels`.

pfeerick, 2022-09-26:
The only outstanding issue after that PR will be separately tracked by #535

closed 2022-10-11 by pfeerick