EdgeTX/edgetx #2110: Widget focus indicator comes up when touching main screen.

MirlynG, 2022-07-02:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Upgraded to 2.7.1 on my TX16S (mk1). All good except for this irritating bug.
When I tap the main model screen in the area of any of the widgets, the widget gets the focus. Touching again does nothing.

On leaving it for about 5 seconds, the highlight flashes a few times and goes away. Touching again simply brings up the highlight again. Clicking the "return" button will remove the highlight at any point.
This behaviour only occurs if trying to use the touch screen in the widgets area. Touching the screen along the top (menu bar), or along the bottom or sides (trim and slider area) works fine and brings up the expected options.
The same thing occurs I inadvertently scroll the jog dial, even just one click. However, if I now click the jog dial, I get the "Full Screen or Widget Settings" options dialogue. Touching either of these takes you to its respective screen. Again, hitting the return button will return you to the main screen but the highlight is still showing. If left it will then flash a few times and disappear as above.
Not a major issue but really annoying.
Very grateful for help to resolve this.

### Expected Behavior

When touching the main screen I expect the menu options (Select Model, Channel Monitor etc) to come up.

### Steps To Reproduce

Touch screen anywhere within the widget area.
Scroll a bit on jog dial.

### Version

2.7.1

### Transmitter

Radiomaster TX16S

### Anything else?

I have uploaded a video which is too large for this environment, but can be viewed here:
https://drive.google.com/file/d/1-LdIZzSXnF5_O3qfCfR2EYi1BxR3LOgt/view?usp=sharing 

or maybe .zip file will work.
[drive-download-20220702T112137Z-001.zip](https://github.com/EdgeTX/edgetx/files/9032929/drive-download-20220702T112137Z-001.zip)

A couple of pictures
[image]
[image]

MirlynG, 2022-07-02:
Apologies. Quoted wrong .zip file. That is for the photos. The link is good though.

pfeerick, 2022-07-02:
Except this is expected behaviour with the current UI (and is not new to 2.7.1 btw). When you tap and hold a widget, the first tap gives it focus, and hold (which has to be within a short period of time) brings up the menu for the widget. The same goes for a double tap - the first tap gives that specific widget focus, and the second tap makes it go full screen. Scrolling the roller does the same as the first tap - which is how you would interact with a widget if you didn't have a touch screen. 

Touching the top bar, or using the roller ENTER is the expected method of bringing up the menu since you're not interacting in the widget area at present. If you have input focus on a widget, you would use the RTN key to cancel it, or wait for it to timeout. 

The problem therein is how to make it so the widget focus isn't so easy to trigger. For the roller, perhaps a long press toggle between giving widget focus or not. But for touch I'm not so sure of the best mechanism - if no second event (tap, hold, slide) bring up the radio menu? 

btw, I'm sure there was another issue somewhat overlapping with this, but I can't find it now. So if someone finds/knows it please link it.

MervDale, 2022-07-03:
Just as point of interest with 2.8 nightly tx16s-74767a3  double tap on the widget does not send it to full screen the only option available now is hold and bring up the widget menu .

pfeerick, 2022-07-03:
Hm, good point. Focus highlight is missing also, which then leads to "why isn't it doing anything" questions until it's documented to be new expected behaviour (which I don't think it is - i.e. think back to a phone UI - do you expect the phone settings menu to come up when you tap between the icons or on a widget? It's no different here, as the widgets can represent an app in it's compact state).

raphaelcoeffic, 2022-07-03:
> Hm, good point. Focus highlight is missing also, which then leads to "why isn't it doing anything" questions until it's documented to be new expected behaviour (which I don't think it is - i.e. [...]

The widget focus indicator is now reserved for keys: use long press on enter to trigger it. You can then select a widget with simple press for context menu.

MirlynG, 2022-07-03:
Thanks for all the comments. As I see it now, the fact that this is not a bug raises a couple of questions.

Do we need a full screen image of the model widget? Why?
Also, why do we need a full screen image of our timers or our RSSI or any other  value for that matter? What is the point?

As for widget settings, it takes just one more click/tap action to get the the widget setting screen, i.e TELE button - setup widgets - tap widget - widget settings. Easy.

It appears to me that these functions are just unnecessary from the main screen and it would be far more beneficial, and much less irritating if they could be disabled so that any tap on the main model screen just brings up the menu.

Reading the comments, it seems there may be some changes in 2.8, but until it is released of course we won't know for sure. I only have one Tx and am reluctant to flash the nightly to see for myself. It may of course be that by the time it is ready for release this behaviour may have been changed. I for one really hope so.

pfeerick, 2022-07-04:
> The widget focus indicator is now reserved for keys: use long press on enter to trigger it. You can then select a widget with simple press for context menu.

Excellent... so just need to fix up the double tap fullscreen behaviour, as that will probably be breaking SoarETX and Robert's toolbox widgets. Unless we've started moving away from the app mode idea we've been working towards?

> Do we need a full screen image of the model widget? Why?
> Also, why do we need a full screen image of our timers or our RSSI or any other value for that matter? What is the point?

Not necessarily. Some people may want ability to show full-screen of model image on demand, but other built-in widgets don't take full advantage of fullscreen (yet?). Timer and values could though (i.e. value could be enlarged, and show both minimum and maximum values as well). But instead look at Timer2, which increases in size when full-screened, and the SoarETX widgets. These in particular are perfect examples of how the 'minimised' state behaves one way, and you can double tap on the widget to access it's full-screen (and interactive) mode. 

This is why I believe nothing should change in as far as being able to access the widget settings with a tap and hold, and then choosing settings (because why do I want to do a button press, and three taps when I can do a tap and hold). All that would be needed is if no second stage is registered (the hold for options, or the second tap for fullscreen), then you are not interacting with a widget, and the radio menu can be shown. 

> Reading the comments, it seems there may be some changes in 2.8, but until it is released of course we won't know for sure.

Not so... you are welcome to try any of the nightly builds to see the state of development... just backup your MODELS and RADIO folders first just incase.

raphaelcoeffic, 2022-07-04:
> Excellent... so just need to fix up the double tap fullscreen behaviour, as that will probably be breaking SoarETX and Robert's toolbox widgets. Unless we've started moving away from the app mode idea we've been working towards?

No, that's right, it's on TODO.

MirlynG, 2022-07-07:
> Not so... you are welcome to try any of the nightly builds to see the state of development... just backup your MODELS and RADIO folders first just in case.

Decided to risk it and to  have flashed the last three nightly builds to my Tx16Ssee what changes have been made so far. Each one loads up OK but each one also goes to Emergency Mode at random and different points when navigating the menus. 

Fortunately been able to re-flash back to 2.7.1 without problems.

Any ideas?

gagarinlg, 2022-07-07:
Can you give us a zip of the model and radio folders from your radio and a description what you did? We can then debug those problems.
I did not have those problems with the last versions I tested

MirlynG, 2022-07-07:
> Hi and thanks for the response.
>
> The problem is extremely random and inconsistent. I would be impossible to describe in words as there are so many different behaviours. Because of this, here are two links which will take you to two videos showing a number of different circumstances where this happens. These do not  necessarily cover every behaviour I have encountered.
>
>
>
> Video#1  https://drive.google.com/file/d/15G7qFaHi6bqnKZsU3wjQ851cNQK1Zfuo/view?usp=sharing
>
> Video#2  https://drive.google.com/file/d/15FumYlMI8FnHAe4f8JxkD2iJP4yZfRpv/view?usp=sharing
>
>
>
> In video#1, you will see me tap a number of icons along the top bar. The problem occurs when I tap the telemetry icon at around 0:16. Interestingly, in video#2, I tap the telemetry icon twice at 0:09 and 0:25 and nothing happens.
>
> In video#2 in Radio Settings, at around 0:51 I tap the Hardware icon and the problem occurs. Later, in screen settings at around 1:17, I tap and hold on the background and nothing happens. But when I swipe up it happens. Later in Radio Settings at around 1:44 in Tools, I swipe up on the background on the right and bingo, once again.
>
>
>
> You can see from these examples that it is as I said, random and inconsistent.
>
>
>
> Hope this all helps.

gagarinlg, 2022-07-07:
☹️
Those are the worst errors.

MirlynG, 2022-07-07:
[MODELS.zip](https://github.com/EdgeTX/edgetx/files/9067528/MODELS.zip)
[RADIO.zip](https://github.com/EdgeTX/edgetx/files/9067529/RADIO.zip)
Not sure these came through

MirlynG, 2022-07-07:
....and thanks again

raphaelcoeffic, 2022-07-09:
@MirlynG we might need your LUA widgets as well. What is the timer widget you are using in the upper left corner? (Looks nice by the way)

MirlynG, 2022-07-09:
Found the basis of the problem. Your referring to the "nice" timer widget I've used got me thinking. So, I removed the widget on one model and replaced it with a standard edgeTX timer. Then I flashed the very latest nightly. Problem gone! That is until I selected a different model which still has the "nice" timer widget on. Issue still there. Reselected the model with a standard timer widget, All functions fine.

I tried to change the timer widget on another model while still on the nightly, but the problem prevented me. So, I re-flashed to 2.7.1, changed a second model timer widget back to standard, re-flashed the nightly, and bingo, I now have two functioning model setups! The others still exhibit the problem though.

Went to the widget developer's site and downloaded their, seemingly, updated widget pack, and flashed to radio. Unfortunately the problem with the models using the "nice" widget still behave badly, but the ones with the standard widget are fine.  Ergo, the issue is with these third party widgets. the widget pack is here:
http://fm2m.jimb40.com/WidgetsPack.html
I would not think that this is really an issue for you guys. I guess The FM2M developer needs to sort it. Perhaps I should contact him and point him to this thread?

gagarinlg, 2022-07-09:
With the timer widget installed and you models I can reproduce this.
Can you please create an extra issue for this?

MirlynG, 2022-07-09:
Done. Title:
https://github.com/EdgeTX/edgetx/issues/2124

pfeerick, 2022-08-03:
Closing this issue as it was actually a duplicate of #2164 which was crossposted. The underlying widget focus indicator doesn't come up any more when you inadvertently roll the scroll wheel, and won't appear if you touch a widget - it only responds to long press for the menu, and double-tap for fullescreen (when that's fixed).

closed 2022-08-03 by pfeerick