EdgeTX/edgetx #2112: Taranis x-lite pro doesn't turn on HappyModel ES24TX Slim Pro ELRS module on boot

pstein29, 2022-07-03:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

When I power on my `Taranis X-lite pro` with a model selected that is configured for `External RF - Mode` set to `CRSF` my `HappyModel ES24TX Slim Pro ELRS` module doesn't turn on.  (the led is not on and the elrsv2.lua script cannot connect to it.

I can get it to turn on by selecting the `External RF - Mode` option and then backing back out (it is already set to CRSF and I leave it that way).  After this the module seems to work as normal.

### Expected Behavior

I would expect the module to be powered and running after boot when the model that is selected during boot is set to use the external module.

I would expect that changing a setting to it's current setting would not cause a change in behavior.

### Steps To Reproduce

1. realize that frsky are jerks and their OpenTX build for `Taranis X-Lite pro` excludes CRSF
2. download and install EdgeTX v2.7.1
3. add model and set `External RF - Mode` to `CRSF` 
4. Flash `HappyModel ES24TX Slim Pro ELRS` and Rx to ExpressLRS 2.5.1
5. reboot Rx and finish model setup
6. reboot Tx

### Version

2.7.1

### Transmitter

FrSky X9 Lite / Lite S

### Anything else?

I am opening an EdgeTX issue because as a troubleshooting step I downgraded to OpenTX (not the Taranis build) and it seems to work fine, the module powers on when I boot (if the correct model is selected) and I don't have to take additional steps to get the module to power on.

pstein29, 2022-07-03:
After submitting this I did some additional searching and found issue: #1224 which appears quite similar. 
The model I was having trouble with was indeed restored...

raphaelcoeffic, 2022-07-04:
> realize that frsky are jerks and their OpenTX build for Taranis X-Lite pro excludes CRSF

🤓😅😬

pstein29, 2022-07-05:
Okay, this may be a misdiagnosis.  While OpenTX would work frequently it didn't power the module every time.  After some extensive searching on rcgroups I've found a PMU update for the radio (frsky x-lite s/pro) [here](https://www.rcgroups.com/forums/showthread.php?3080337-FrSky-Taranis-X-Lite-S-and-Pro-X9-Lite-S-R9M-Lite-and-Pro-MPM-Lite/page242#post43747003)  
Flashing this file has caused OpenTX to boot the module every time.  (also the module isn't powered when charging the radio via usb).

While I haven't re-flashed EdgeTX I'm reasonably confident that this was the root cause of my issue.  I'm going to close this issue given the above development.

closed 2022-07-05 by pstein29

pfeerick, 2022-07-06:
Glad to hear it's working for you now. Please let us know how you go when you try ETX out next.

danflyingrobot, 2024-05-16:
> Okay, this may be a misdiagnosis. While OpenTX would work frequently it didn't power the module every time. After some extensive searching on rcgroups I've found a PMU update for the radio (frsky x-lite s/pro) [here](https://www.rcgroups.com/forums/showthread.php?3080337-FrSky-Taranis-X-Lite-S-and-Pro-X9-Lite-S-R9M-Lite-and-Pro-MPM-Lite/page242#post43747003) Flashing this file has caused OpenTX to boot(power) the module every time. (also the module isn't powered when charging the radio via usb).
> 
> While I haven't re-flashed EdgeTX I'm reasonably confident that this was the root cause of my issue. I'm going to close this issue given the above development.

How is the radio treating you? Any battery capacity issues running elrs?

Camo55, 2024-05-17:
I noticed sometimes my jumper elrs module would get stuck in a boot loop, though I switched to 400k baudrate and that seems to have mostly fixed that issue

danflyingrobot, 2024-05-17:
> I noticed sometimes my jumper elrs module would get stuck in a boot loop (since updatung to elrs 3.4.0) though I switched to 400k baudrate and that seems to have mostly fixed that issue. (Also frsky xlite pro transmitter)

Probably the module then