EdgeTX/edgetx #2116: Silent mode warning

Eldenroot, 2022-07-04:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Add a warning during start-up that the sound is not enabled (the volume is muted)

### Describe the solution you'd like

Add a new option for this.

pfeerick, 2022-07-15:
Do you mean as in you've turned the volume down (specifically, how - via the volume slider, wav vol slider, SF?), rather than set the alarms mode to "Quiet"? As there is already an alarm warning and option if you have it set to Quiet.

Eldenroot, 2022-08-20:
Yes, just to know that volume is muted during booting phase of the radio. Because it could be dangerous to fly and wait for any voice/beep warning but nothing will come...