EdgeTX/edgetx #2118: Haptic patterns

Eldenroot, 2022-07-04:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Would be nice to add a custom haptic patterns and use them as a new way for warning a user.

For example - in a special/global function menu, in telemetry, global warning etc.

### Describe the solution you'd like

Add a new settings for this, maybe a possibility to a generate any files which contain a pattern for haptic vibration.

tekikai, 2022-07-23:
You could always write your own haptic patterns in LUA but I agree that a dedicated option via file or even interface would be great.
I currently wrote my own very simple haptic feedback function and integrated it via a **logic switch** and **special functions**.

Eldenroot, 2023-01-25:
It could be done this way:

- Strength of haptic feedback
- haptic pattern:
* "-"
* "- -"
* "- - -"
* "- - - -"
* "- - - - -"
* "- . - . -"
* "."

. = short 
- = long vibration

JimB40, 2023-01-26:
". --" for EM ;)

Eldenroot, 2023-01-26:
Yeah, there could be more ;) Lets gather the feedback!

pfeerick, 2023-01-26:
No... EM would be `... --- ...` 😆

Eldenroot, 2023-02-12:
Maybe we do not need custom patterns... there would be better add lets say 5-8 preconfigured :)

DAMskill, 2024-07-07:
> You could always write your own haptic patterns in LUA but I agree that a dedicated option via file or even interface would be great.
> I currently wrote my own very simple haptic feedback function and integrated it via a **logic switch** and **special functions**.

How did you manage to do it in lua?
I never programmed lua.

pfeerick, 2024-07-07:
Would probably be a [Function script](https://luadoc.edgetx.org/part_i_-_script_type_overview/function_scripts) that is using a couple of [`playHaptic()`](https://luadoc.edgetx.org/part_iii_-_opentx_lua_api_reference/general-functions-less-than-greater-than-luadoc-begin-general/playhaptic) commands.