EdgeTX/edgetx #2124: Tx16s Edge 2.8 goes to Emergency Mode randomly when navigating if FM2M widget pack is used.

MirlynG, 2022-07-09:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

When using the custom timer included in the FM2M widget pack  (see link below) with EdgeTX 2.8 nightlys, the tx goes to emergency mode randomly and inconsistently when navigating a model which uses this timer. This behaviour can be seen in the videos linked below.
If this timer is replaced with a standard EdgeTX one, the tx performs as it should.

FM2M widget pack: http://fm2m.jimb40.com/WidgetsPack.html

Video links showing behaviour:
Video#1  https://drive.google.com/file/d/15G7qFaHi6bqnKZsU3wjQ851cNQK1Zfuo/view?usp=sharing
Video#2  https://drive.google.com/file/d/15FumYlMI8FnHAe4f8JxkD2iJP4yZfRpv/view?usp=sharing

### Expected Behavior

Tx should not shut down

### Steps To Reproduce

Install FM2M widget pack and create a model with the included timer.
Navigate through the menus. Sooner or later the tx will switch to emergency mode as shown in the videos detailed above.

### Version

Other (Please specify below)

### Transmitter

Radiomaster TX16S

### Anything else?

Only happens with v2.8

gagarinlg, 2022-07-09:
[MODELS.zip](https://github.com/EdgeTX/edgetx/files/9077944/MODELS.zip)
[RADIO.zip](https://github.com/EdgeTX/edgetx/files/9077945/RADIO.zip)

Added the files uploaded to #2110 , as just using the timer widget did not trigger the problem, a model from this MODELS:zip had to be used.
LUA scripts just should not be able to crash the radio, regardless of any errors in the LUA scripts.

gagarinlg, 2022-07-11:
When switching tabs in a tabbed screen, it may happen that main screen lua widgets get called and drawn, while they should not draw on the screen because they are not visible. 
Somehow this leads to a processor hardfault sometimes.m I have not been able to track it down to a specific call, yet.

gagarinlg, 2022-07-22:
More information:
Every time an input event is handled lua widgets are called, even when the main screen is not visible. Then bad things can happen, when the script draws onto the screen

raphaelcoeffic, 2022-07-23:
@gagarinlg did you manage to reproduce this on simu?

gagarinlg, 2022-07-23:
No, I made a little Lua script but it did not crash and the original ones do not work, because the Lua binaries won't work, maybe because I compile for 64 bit.

gagarinlg, 2022-07-23:
I think I got it in simu.
`void BitmapBuffer::drawAnnulusSector(coord_t x, coord_t y, coord_t internalRadius, coord_t externalRadius, int startAngle, int endAngle, LcdFlags flags)
{
  if (endAngle == startAngle) {
    endAngle += 1;
  }

  Slope startSlope(startAngle);
  Slope endSlope(endAngle);

  pixel_t color = COLOR_VAL(flags);
  APPLY_OFFSET();

  coord_t internalDist = internalRadius * internalRadius;
  coord_t externalDist = externalRadius * externalRadius;

  DMAWait();
  for (int y1 = 0; y1 <= externalRadius; y1++) {
    for (int x1 = 0; x1 <= externalRadius; x1++) {
      auto dist = x1 * x1 + y1 * y1;
      if (dist >= internalDist && dist <= externalDist) {
        Slope slope(false, x1 == 0 ? 99000 : y1 * 100 / x1);
        if (slope.isBetween(startSlope, endSlope))
          drawPixelAbs(x + x1, y - y1, color);
        if (slope.invertVertical().isBetween(startSlope, endSlope))
          drawPixelAbs(x + x1, y + y1, color);
        if (slope.invertHorizontal().isBetween(startSlope, endSlope))
          drawPixelAbs(x - x1, y + y1, color);
        if (slope.invertVertical().isBetween(startSlope, endSlope))
          drawPixelAbs(x - x1, y - y1, color);
      }
    }
  }
}`

Depending on the parameter, drawPixelAbs might get called with negative x and/or y coordinates

`    inline void drawPixelAbs(coord_t x, coord_t y, pixel_t value)
    {
      pixel_t * p = getPixelPtrAbs(x, y);
      drawPixel(p, value);
    }
`
then getPixelPtrAbs is called with negative coordinates
` 
inline pixel_t * getPixelPtrAbs(coord_t x, coord_t y)
    {
#if defined(LCD_VERTICAL_INVERT) && 0
      x = _width - x - 1;
      y = _height - y - 1;
#endif
      return &data[y * _width + x];
    }
`

Now you have a pointer to somewhere, but not your buffer. And then the pixel is drawn to that somewhere.

This is defenitely a bug, but that the lua widgets get drawn, when the main screen is not visible is also a bug, in my opinion.

gagarinlg, 2022-07-23:
To reproduce, just add the GaugeRotary widget to the main screen and bind it to a running timer. Then enter the radio setup page and scroll around. The creash should happen within seconds.

gagarinlg, 2022-07-23:
There is a check against it in the code, but with a slight logic error.

```
    inline void drawPixel(pixel_t * p, pixel_t value)
    {
      if (data && (data <= p || p < data_end)) {
        *p = value;
      }
#if defined(DEBUG)
      else if (!leakReported) {
        leakReported = true;
        TRACE("BitmapBuffer(%p).drawPixel(): buffer overrun, data: %p, written at: %p", this, data, p);
      }
#endif
    }
```

raphaelcoeffic, 2022-07-23:
@gagarinlg thx! This is definitely a job for the address sanitizer.

gagarinlg, 2022-07-23:
The underlying reason is, that the drawAnnulus does just draw wherever it wants and not respect the draw context.
I started to fix this by using the draw_annulus function of the lvgl draw context, but it is not done yet.

pfeerick, 2022-07-24:
I've been seeing those... particularities (I'm being polite 😝) around the drawAnnulus stuff also - i.e. as I noticed if I use the clock widget from that pack, that it seemed to draw on the model setup -> telemetry sensors page when you scroll to the botom... which clearly it shouldn't (on 2.7.1).... and I think perhaps there is some sort of slowdown also as the transmitter just feels a bit more sluggish as soon as you add the clock widget, and definitely so with the GaugeRotary widget. Just hadn't gotten around to trying to pin it down a bit more to make an issue, and hadn't seen EMs from it yet...

gagarinlg, 2022-07-24:
EM is only happening with 2.8 as there are changes to the surrounding drawing code.
What I am still wondering is, if we really need to draw the Lua widgets, when they are invisible and why lvgl does not detect that they are hidden.

raphaelcoeffic, 2022-07-24:
@gagarinlg we should not. There are a few aspects, which we should discuss separately.

gagarinlg, 2022-07-24:
The crash is fixed with [libopenui#55](https://github.com/EdgeTX/libopenui/pull/55), but not the real problem.
[libopenui#56](https://github.com/EdgeTX/libopenui/pull/56) fixes the real problem for drawAnnulus and the circles look nicer with the change.
At least the drawCircle function also needs to be fixed, LVGL does not provide an equally efficient alternative, though.
To stop the creashes libopenui#55 has to be merges and EdgeTX/main has to be updated to use the then modified libopenui.

gagarinlg, 2022-07-28:
Current nightlies should not crash anymore

MirlynG, 2022-07-29:
You guys are the best!. Just flashed latest nightly and certainly, the problem seems to be solved. Thankyou so much.

However, during my testing I found something in failsafe setting, which I am pretty sure is not related to the FM2M widgets.

I see it has already been reported here: Failsafe Error #2162

gagarinlg, 2022-07-29:
> You guys are the best!. Just flashed latest nightly and certainly, the problem seems to be solved. Thankyou so much.
> 
half way fixed currently, but at least it does not crash anymore

> However, during my testing I found something in failsafe setting, which I am pretty sure is not related to the FM2M widgets.
> 
> I see it has already been reported here: Failsafe Error #2162

It just got fixed, I think. See PR #2168

pfeerick, 2022-08-03:
Closing this issue as the EM has been resolved, and https://github.com/EdgeTX/libopenui/pull/56 will take this further and fix the underlying issue.

closed 2022-08-03 by pfeerick