EdgeTX/edgetx #2130: Model and radio settings disappear

EDeuchar, 2022-07-15:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Some times when I turn off my radio then turn it back on all the model data disappears, or becomes corrupted. It also reverts all radio data to the absolute default, it doesn't even have inputs. When I try to read the radio data in companion it shows that some of the data lines have unknown characters. I am not sure if it's a sd card issue or a firmware issue, it only became an issue when I upgraded to 2.7.1.

### Expected Behavior

Data should remain on power cycle.

### Steps To Reproduce

1. Turn radio on.
2. Set up model data.
3. Turn radio off.
4. May need to repeat.

### Version

2.7.1

### Transmitter

Radiomaster TX16S

gagarinlg, 2022-07-15:
This might be fixed as soon as #1481 is merged.
You should try to use a different SD card, maybe it is not good anymore.

gagarinlg, 2022-09-12:
Can you please test with the current nightlies?

gagarinlg, 2023-03-18:
With 2.8 this should be fixed, so this issue is closed now.
If it happens again, please open a new issue

closed 2023-03-18 by gagarinlg