EdgeTX/edgetx #2131: wrong mixer values

tonims1, 2022-07-16:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

in the mixers part, the weight and offset allow values ​​from -500 to 500 ,
Edgetx 2.8 main

### Expected Behavior

They should have values ​​between -100 to 100

### Steps To Reproduce

open the mixer screen put maximum and minimum value in weight and offset

### Version

Other (Please specify below)

### Transmitter

Radiomaster TX16S

### Anything else?

[image]

pfeerick, 2022-07-16:
Actually, no, it looks like you *should* be able to set it to +/- 500% for both... don't ask me why though...

i.e. you can do this with 2.7.x, and also with OpenTX... so it's not accidental / a bug. Will need to investigate further / have someone more knowledgeable comment on the *why*. 

[image]

pfeerick, 2022-07-16:
Ok, this is most definitely not accidental or a bug... i.e. OTX University mentions this quite behaviour clearly:
> This is the weight (in %) to be applied to the input to determine how much of the input channel gets into the output from the mixer. The default is 100% for all lines but 0% to 500% can be entered here to achieve the desired ratio of different inputs into one output.  Once the proper ratios are achieved, a Mixer channel output that exceeds 100% is clipped, or stopped, at 100% to be passed on to the Servos screen where the final control throw is determined with travel limits and servo reversals are defined.

http://open-txu.org/home/undergraduate-courses/basic-airplane-set-up-with-taranis-2/mixer-613/

Also, I see mention of it being used like this by one of the OTX devs:
> If you want the full travel on a servo with less stick travel you just increase the mixer weight. If you set 500% then it'll reach full with 1/5th of the travel.

I then see it used in some rather heady mixing diagram:
https://www.mvcpegasus.nl/wp-content/uploads/OpenTX_Documents/OpenTX%20Diagram%20-%20Mixes.pdf

Given the value is clipped to the range specified in the outputs page, it seems like this is there to allow things like the aforementioned increase (or decrease) of travel range with more or less input, and other fancy programming with curves, expo, GVs, etc.

closed 2022-07-16 by pfeerick