EdgeTX/edgetx #2141: Have YAML file on retain comments

rdeanchurch, 2022-07-18:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Currently comments are stripped off and discarded.  
This means comments can not be used to document or as an aid to editing the YAML files for modelxx.yml.

### Describe the solution you'd like

Allow # comments to be retained after being processed by the Tx O.S..

### Additional context

# timers...Begin...This block of YAML can set up three timers for a model by copying it to a
#   new model after the 'header: object so all models have a consistent set of timers
timers:
  0:     # TIMER 1
    swtch: SF0  # Sw SF down is used as a Throttle safety, up is armed
    mode: ON
    name: Flight
    minuteBeep: 0
    countdownBeep: 0
    start: 0
    persistent: 0
    countdownStart: 0
    value: 0
  1:
    swtch: SF0
    mode: THR
    name: Throttle
    minuteBeep: 0
    countdownBeep: 0
    start: 0
    persistent: 0
    countdownStart: 0
    value: 0
  2:
    swtch: NONE
    mode: ON
    name: Toal
    minuteBeep: 0
    countdownBeep: 0
    start: 0
    persistent: 1
    countdownStart: 0
    value: 0
# end  timers block

pfeerick, 2022-07-18:
For completeness, this stems from the discussion in #2140. 

It would also need Companion support, as a read and write cycle from Companion will also strip anything that is not model/radio data.

elecpower, 2022-07-18:
Need to consider impact/limitations for B&W radios due to their limited ram for parsing/storing/writing the comments

gagarinlg, 2022-07-22:
Since the files are regenerated every time they are written in the order the firmware thinks is right, I do not see this happen.
We would need to store a link between the comment and a config option. Which one should we take? The option before or after the comment?

pfeerick, 2022-07-22:
> Since the files are regenerated every time they are written in the order the firmware thinks is right, I do not see this happen.

I had this feeling that would be the case... Lets take break this down further for both me and other end users... if my understanding is right it is read from file into memory, and if a setting is changed in memory, it's then taken from memory back to file... so since comments are not settings, they don't exist in memory, thus wouldn't be read out/written back... sound about right? And since this would consume tones of memory (which we don't have) it's not practical on lower memory targets.

gagarinlg, 2023-03-18:
Stale and I also my opinion has not changed.
I do not see this happening, especially not on the B&W radios with limited memory

closed 2023-03-18 by gagarinlg