EdgeTX/edgetx #2142: UIX Inconsistency

MartinHugh, 2022-07-18:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

If I want to select a model, from the Select Model page, the model to be selected is delineated in a solid frame. The other models are not delineated so it is obvious which one I am selecting.

If I go to edit screen widgets, I need to select one to bring up the submenu to select "Widget Settings".
However, all the widgets which are not selected are each delineated in a solid frame. The one to be selected is in a faint dotted white frame.

I cant tell you how often when I am at the flying field trying to adjust a widget I get this wrong. This is particularly likely when there are only two widgets set and one is framed and one isnt.

### Describe the solution you'd like

Any UI item which is being nominated for selection should be consistent across screens/pages and should be easy to see.  In this case it means that it should be delineated in a solid border/frame.

### Describe alternatives you've considered

N/A

### Additional context

https://github.com/EdgeTX/edgetx/discussions/2137

MartinHugh, 2022-12-13:
Is this planned to be in the next release?  Enjoying 2.8 very much, but notice that this issue has not been implemented so far.

philmoz, 2024-08-03:
Implemented in 2.11.

closed 2024-08-03 by pfeerick