EdgeTX/edgetx #2143: Selecting Channels=>Failsafe button in the Failsafe Settig menu crashes W10 Companion

mha1, 2022-07-19:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Companion

### Current Behavior

Selecting `Channels=>Failsafe` button in the Failsafe Settig menu crashes Companion.

### Expected Behavior

Companion not crashing, current Inputs are stored as failsafe values

### Steps To Reproduce

1. Open Companion and create new model
2. Open newly created model
3.  In the Setup tab select `Internal Radio`  `System Multi` and select `Multi Radio Protocol` `HoTT ` (alphabetically sorted list would be much nicer)
4. Click `Simulate`
5. In the simulated radio screen select `Model Setup` and click the `Internal RF button`
6. Scroll down to `Failsafe mode` and select `Custom` the click on the `Set` button
7. In the FAILSAFE SETTINGS (not all capitals would be nicer) screen click on the `Channels=>Failsafe` button
8. Observe Companion crashing

### Version

Other (Please specify below)

### Transmitter

Radiomaster TX16S

### Anything else?

Main branch - commit 18db00d8

I haven't checked radio firmware, fault might be present in firmware too.

mha1, 2022-07-28:
for radio see https://github.com/EdgeTX/edgetx/issues/2162

@elecpower @pfeerick can we take this out of triage please

mha1, 2022-07-28:
@pfeerick Thank you

pfeerick, 2022-07-31:
Since this fix didn't touch companion specific code, it won't be visible there until the next companion build is triggered.

PR #2170 (2022-07-31): fix: type cast for failsafe channel update
Fixes #2143, #2162
closed 2022-07-31 by pfeerick