EdgeTX/edgetx #2147: X12S ISRM/ACCESS missing flash RX by int. OTA option on .frsk files

rburrow87, 2022-07-20:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Only getting the options to flash internal or external module when trying to flash a receiver .frsk file.

### Expected Behavior

With internal module set to ISRM, there should be a `Flash RX by int. OTA` option. In OpenTX, it can tell if it's firmware for the internal ISRM module and offer to flash internal module.

### Steps To Reproduce

With the internal module set to ISRM, go to flash a .frsk receiver firmware file.

### Version

2.7.1

### Transmitter

FrSky X12

### Anything else?

This is present in both 2.7.1 official build and tonight's main nightly https://github.com/EdgeTX/edgetx/commit/f31f06b344471106512fdfbd2a2e6a273ea63a5a

EdgeTX 2.7.1 official build:
[image]

OpenTX 2.3.15:
[image]

offer-shmuely, 2022-08-14:
also on X10s access, there is no option of OTA
only the "internal module" and "external module"

elekSENDER, 2022-08-15:
Same issue on a X10s (not express) with an upgrade-ISRM-module. 
Just "internal module" and "S.Port device"

raphaelcoeffic, 2022-08-15:
The main issue here is: I have no way of testing this.... would have to buy an OTA receiver extra for this. Maybe someone with the hardware could take care? I'm a bit reluctant to buy this whereby I don't use it myself at all (D16 still, and moving all planes to ELRS right now).

gagarinlg, 2022-08-15:
> The main issue here is: I have no way of testing this.... would have to buy an OTA receiver extra for this. Maybe someone with the hardware could take care? I'm a bit reluctant to buy this whereby I don't use it myself at all (D16 still, and moving all planes to ELRS right now).

Just by it on EdgeTX money. It is testing equipment for a supported hardware.

raphaelcoeffic, 2022-08-15:
@gagarinlg right, just ordered a R9M 2019 (30€) and a R9 Slim+ OTA, which should fit the purpose.

elekSENDER, 2022-08-15:
I can do some test for ACCESS (LBT)-transmission on a X10s (not express) with an upgrade-ISRM-module and an ARCHER-RX.
How do I know when it is ready for testing?
And where can i download the firmware, is this the nightly version?

raphaelcoeffic, 2022-08-15:
> I can do some test for ACCESS (LBT)-transmission on a X10s (not express) with an upgrade-ISRM-module and an ARCHER-RX. How do I know when it is ready for testing? And where can i download the firmware, is this the nightly version?

Not so fast.... step (1) (ordering some hardware) has just been completed.

(2) Then I need to check if I can make it work, then open a PR, and the PR gets tested, and then it gets merged.

(3) This is when it gets into nightly builds.

raphaelcoeffic, 2022-08-15:
Frankly, this is why I strongly prefer ELRS: OTA just works out of the box via Wifi without any implementation in EdgeTX ;-)
Plus, all we have is the code in OpenTX, no description or documentation from FrSky....

elekSENDER, 2022-08-15:
understood. ;-)
Please don´t get me wrong, I do not want to make any pressure.
I´m just the guy who can test something and say yes, works for me or no, and this happend ...
Unfortunatly I´m not familiar with the processes of github and i only wanted to know how can i see that it does make sense to download the nightly for a special issue.

pfeerick, 2022-08-15:
@elekSENDER Once there is a PR available for testing, you'll be able to download the firmware for it from the Checks tab of the PR, or via EdgeTX Buddy (if you have the expert mode setting on).

fabmce, 2022-09-23:
Bonjour à tous
Je croyais que j'étais seul  dans ce cas , mais je vois qu'il commence à y avoir du monde .
effectivement le problème existe bien car il est impossible de faire une OTA sur un RX archer depuis EDGETX , j'ai essayer sur differents rx de la gamme archer mais la mise à jour par OTA ne figure pas dans l'onglet . je suis donc revenu vers OPENTX en attendant qu'une mise à jour vienne pour EDGTX afin de pouvoir flasher les rx archer ... Wait and see

ParkerEde, 2022-09-23:
Hi,  I have X10S  Express and Archer  RX, so I‘m able to test it in 2.8.0 too.

raphaelcoeffic, 2022-09-24:
And I have all the hardware now ready, so I should be able to try this out very soon.

ParkerEde, 2022-10-18:
is this ticket a case for v. 2.8 or what do you think about it?

raphaelcoeffic, 2022-10-18:
No, this requires a lot of code to be ported. A little while ago, I asked for contributions regarding these FrSky modules. For my part, I don‘t use all this anymore, so I would prefer to find a maintainer who is really using it himself. Makes just more sense.

pfeerick, 2022-10-19:
> is this ticket a case for v. 2.8 or what do you think about it?

For 2.8. not a chance... the feature change window has close well and truly ;) It was only finishing up features slated for 2.8 which were still WiP,  and then bugfixes, regression or minor UI/translation changes once we entered RC. 

Frankly speaking, I would not be surprised to see support dropped for Frsky in the future if we don't have users who can also help with development for it or collaboration from Frsky. It just does not seem fair to other manufacturers who have pro-actively established avenues of communication and provide both financial and hardware support as needed. This is just my personal opinion, it does not mean it's happening. I would be sad to see it happen, as I do have Frsky gear myself, and do use it still, but that is just the reality of the situation at present. Then again, if we come to realise a significant number of users are using ETX with Frsky radios, that would justify using a proportional amount of the funds raised to support those radios.

ParkerEde, 2022-10-19:
@raphaelcoeffic 
> And I have all the hardware now ready, so I should be able to try this out very soon.

> No, this requires a lot of code to be ported. A little while ago, I asked for contributions regarding these FrSky modules. For my part, I don‘t use all this anymore, so I would prefer to find a maintainer who is really using it himself. Makes just more sense.

I first understood you to say that you wanted to look at it soon, now that you have the hardware. Do I now understand you correctly that you will not do it after all?

pfeerick, 2022-10-19:
> I am surprised that this ticket is a problem, since it is already included and functional in OTX 2.3.14

Yeah, well, I'm still surprised OpenTX doesn't have touch support. 🙃 Something to keep in mind is that there have been a lot of re-writes to underlying code, and that the re-write of the OTX 2.4 UI for was not maintained in parallel with OpenTX 2.3 development (meaing there was a lot of stuff missing and buggy when it was forked to EdgeTX), so just because it worked on OTX doesn't mean it will work here. For B&W radios we can generally pull code from OTX for a lot of things, but the colorlcd code has now been heavily re-written again with the port to LVGL, and the underlying hardware drivers are also being replaced bit by bit in each version of EdgeTX, to allow for newer hardware platforms to be supported. I'm just saying be patient - Rapheal will probably work on this if no one else steps up, but I don't think it will be a priority just atm, especially as it is of no use to him as a non-ACCESS user.

ParkerEde, 2022-10-19:
ah okay. Thank you for the explanation. I did not know all this from the background knowledge

ParkerEde, 2022-10-19:
If porting code at any point is just busywork (without being able to develop properly) let me know, I'm happy to support.

raphaelcoeffic, 2022-10-19:
> porting code at any point is just busywork

if that would have been the case, I’d have done it already. But there is quite more to this, mainly due to the full paradigm changes in the way the UI works now. Also, it requires quite some tests on hardware to get it right, which slows down the development.

last but not least: you can already flash your receivers today, just not over the air.

TakeToTheSky, 2023-02-17:
A few of us have successfully converted our Radiomaster TX16S radios to FRSKY ACCESS by installing an ISRM RF board.  I'm grateful for all the amazing work that the devs do and appreciate that this is not a high priority issue.  Is there any progress on the implementation of the OTA feature?
[image]

PR #3687 (2023-06-30): Add RX OTA flash for frsky X10s & X12s.
Fixes #2147 

Summary of changes:
The OTA flash code are basically copied from stdlcd, rewrite UI part based on the new implementation for color screens.

The UI part cost me some time. My finding is to close dialogs & menus programmatically, they needs to be deleteLater() in stack order, from latest widget to older widget. If the order is wrong the UI will crash immediately. After fix the order of deleteLater(), the UI works fine.

I run several tests in my X10s, click no and yes, flash by OTA several times. Seems working well in my device. It should al [... cut]
closed 2023-06-30 by pfeerick