EdgeTX/edgetx #2150: Not found Chinese languge for Jumper T-Lite

icode, 2022-07-21:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Not found Chinese languge for Jumper T-Lite.

### Describe the solution you'd like

Why? And how to found Jumper T-Lite Chinese version?

pfeerick, 2022-07-22:
I believe TW and CN languages for B&W have been unsupported since EdgeTX 2.4 due to the fonts needed for those languages being too large for the flash storage for those radios, but will have to double check on that, as there has been some changes around font handling, sizes and also string handling, so it may be possible now.

icode, 2022-07-22:
> I believe TW and CN languages for B&W have been unsupported since EdgeTX 2.4 due to the fonts needed for those languages being too large for the flash storage for those radios, but will have to [...]

Maybe the clipped font library can fit completely

YI-BOYANG, 2022-07-23:
屏幕太小了，还不如买个tx16s

pfeerick, 2022-07-23:
That may be so but some people like/prefer it. Other transmitters like the Zorro and TX12 are also the same resolution screen (but physically larger), and this same limitation applies to them, so it would be nice to finally support them.

arthurson, 2022-08-03:
i also faced this problem, the only solution at the moment is to change another colour lcd controller like rm tx16s if you really need chinese menu.

closed 2025-02-22 by github-actions[bot]