EdgeTX/edgetx #2156: ExpressLRS telemetry on simulator

offer-shmuely, 2022-07-23:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Companion Simulator

Now the expressLRS is more and more integrative in edgeTx,
is is very hard to debug lua script that support expressLRS 

### Describe the solution you'd like

enhance the "telemetry" screen on the companion simulator with the elrs default telemetry
that will be easier to build and debug lua script that support expressLRS as well

add the following:
[image]

add them to here:
[image]

ggoraa, 2023-06-12:
This is a good suggestion! afaik this telemetry simulator was not updated in a long time, that's why it doesn't have a lot of fields

rotorman, 2024-08-02:
https://edgetx.org/contest_crsf_telem_simulation/

PR #5410 (2024-08-20): feat(cpn): support more than just FrSky S.Port telemetry simulation
Fixes #4983
Fixes #2156 

Summary of changes:

1. Replace the current telemetry sensor UI dropdowns to choose a telemetry source for both the internal and external module.
2. Refactor the current FrSky S.port telemetry implementation to allow it to have separate telemetry for both internal and external modules at the same time
3. Update the simulator interface to accept different telemetry protocols
4. Implement a CRSF telemetry source and associated UI
5. Rejigger the way the log replay controller works so that it can work wi [... cut]
closed 2024-08-20 by pfeerick