EdgeTX/edgetx #2158: Settings do not get saved over a reboot (Internal TX setting, dead zone setting) on 2.7.1 on a NV14

tekikai, 2022-07-23:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

I usually set the internal TX function to off while using the simulator and also I set the deadzone to 0 (from 4).
These settings get reverted (TX to ON, deadzone to 4) on reboot / restart of the Nirvana NV14 on EdgeTX 2.7.1

### Expected Behavior

I would expect those settings to remain saved over a reboot / restart.

### Steps To Reproduce

1. Go to to: Radio Settings --> Hardware tab
2. Set "Internal RF / internal module" to OFF, set deadzone (under 'sticks') to 0
3. Switch the radio off and on again.
4. Go back to the Hardware tab, settings are reverted to their default states.

### Version

2.7.1

### Transmitter

FlySky NV14

pfeerick, 2022-08-03:
I can reproduce this on latest nightly... a0770beb5f8616e2f14f2828de12b3ff8b1f3397. The dead stick value not saving is specific to `0` - i.e. if it is set to other values, like 2, it is saved. So both are most likely related to a `0` value. 

btw, for simulator, why not a separate model with the internal RF off? Then you can have all your mixes/switches and stuff all configured for simulator as well.

tekikai, 2022-08-04:
Ah, thanks for looking into it, fingers crossed for an easy fix! 
Re simulator: I have a few simulator profiles now (one for Liftoff, one for DCL and one for Elite) but I didn't know if the Internal RF setting in the radio settings gets overwritten by the model settings or not (which I suppose it does according to your comment?).

pfeerick, 2022-08-04:
There are two different parts to this... there is the "what type is the internal module" setting (which is what you are changing in the hardware setttings, and is getting reset to the firmware default if 'OFF'), and there "is the internal module enabled or disabled" setting, which is what is in the model settings. In other words, if you turn it off in the model settings, it is off, because the _type_ of internal module is of no importance at that point.

tekikai, 2023-01-17:
Hi, updated to the latest official Firmware (2.8.0), the problem with the deadzone settings not getting saved over a reboot still remains. Could the issue be reopened or should I open a new one?
Regards, Tekikai

pfeerick, 2023-02-26:
What does it do - by any chance does it revert back to something like 2?

gagarinlg, 2023-09-18:
@philmoz didn't you fix this?

philmoz, 2023-09-18:
No I didn't do anything to fix this.

I would suggest simply removing the code that overwrites the dead zone in the postRadioSettingsLoad function.
The dead zone is set to the default value when the radio.yml file is first created (generalDefaults function) - if the user sets the value to 0 we should not be updating it after loading the radio.yml settings.

PR #4064 (2023-11-11): fix(color): Improvements to NV14/EL18 dead zone logic.
Fixes #2158
closed 2023-11-11 by pfeerick