EdgeTX/edgetx #2161: Fullscreen Keyboard

MadMonkey87, 2022-07-24:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

The touch screen keyboard is a nice feature, but unfortunately it almost always covers the input field so it is not visible what you're typing. This is especially annoying as depending on the context, the responsiveness is not very high so it is easy to miss a few characters.

### Describe the solution you'd like

Most modern touch friendly interfaces rearange the layout s.t the input is always visible but I think this is very hard to implement for edge tx. Instead I suggest to simply make the keyboard fullscreen and use the upper part to display the typed text.

### Describe alternatives you've considered

Use the scroll wheel but this is not as comfortable as the touch keyboard

gagarinlg, 2022-07-24:
Did you test the current nightlies? There is a major redesign ongoing.

MadMonkey87, 2022-07-29:
Just tried it, looks way better! It actually seems to move the input box so it is always visible what you're typing. Great work!

closed 2022-07-29 by MadMonkey87