EdgeTX/edgetx #2165: EEProm conversion proces hangs on a X9E Taranis

Bratski, 2022-07-28:
I tried to migrate from Opentx 2.3.14 to EdgeTX 2.7.1 on a FRSKY Taranis X9E. The System hangs when the EEPROM is being converted. I tried different SD Cards, it didn't solve the problem.

pfeerick, 2022-11-26:
The M9 are the Hall gimbals? If so, there is a specific X9E-hall firmware build now, so you will be able to use that. I'm not sure about the EEPROM hang though. You may be able to work around this by using OpenTX Companion to read the radio, saving the file as a `.otx` file, and then open that in EdgeTX Companion, and writing the settings back to the radio.

pfeerick, 2023-04-06:
Hopefully, this was fixed by #3232, thus in v2.8.2 (and 2.9.x when released). Please let us know if that is no the case and the issue can be re-opened if needed.

closed 2023-04-06 by pfeerick