EdgeTX/edgetx #2166: Models not viewable after copying from OpenTX

mactac, 2022-07-28:
I have a new Radiomaster TX16S radio with EdgeTX.   I tried copying my models over from my Taranis (OpenTX), here was my process:

- Back up models from Taranis using openTX companion, save to disk
- Plug in TX16S with EdgeTX
- Load Taranis models back into companion
- Convert
- Drag & drop (or use right click-copy) models over to my TX16s
- Write models & settings
- unplug radio from computer

Now, my radio does not seems to have any of the copied models on it.  If I plug it back into companion, and "read models and settings", it shows all the copied models properly, just as if they were on the radio.   They just don't show up on the radio itself!

A couple of things I noticed:
-On my openTX radio, there is no "RADIO" folder, although I keep seeing reference to one in online info.  There is one on my EdgeTX radio
- The files that get copied to my new radio are .bin files, though EdgeTX does not seem to use those, instead .yml files.  Should my bin files been converted to yml files?    If so, companion didn't seem to do it
- Another weird thing is that I have about 40 models on my taranis, although the MODELS folder only seems to show 8.  Where are all the other models hiding?

I've tried to copy the models about 30 different times, in different ways, and I've tried re-flashing the TX16S a ton of times, and tried writing the SD card manually and with the tool.  I've tried creating the MODELS and RADIO folders manually, but nothing seems to help.  I even tried upgrading my Taranis to the most recent openTX just in case it needed that before converting.

What am I missing here?

raphaelcoeffic, 2022-07-31:
The issue is probably the models list, which is stored in `/RADIO`, and possibly not copied / read when `/RADIO/radio.yml` is not present. Companion does not copy it if not explicitly asked to.

The trick is the following:
- start the TX16S at least once with EdgeTX
- import the data into Companion
- use that file to copy the taranis models into
- export the data back to TX16S from Companion

RomanLut, 2022-11-29:
Jumpter T-Lite (bw screen) does not have /RADIO folder. /MODELS folder is empty.  I think all settings are kept in eeprom.
One can "backup" models to create .bin files in /MODELS foder on SD card, but they will not be converted.
There is no "SD Card needs conversion" step after upgrade.

Solution is to Save models to *.otx file using OpenTX companion before upgrade,
then load it in EdgeTX companion and write to radio after upgrade.

I had to revert back to OpenTX to save *.otx file...

pfeerick, 2022-11-29:
Correct - this issue is focused on the TX16S - which stores all model and radio settings on the SD card (as do all other colorlcd radios). So if for some reason the firmware did not automatically import the models from EEPROM when migrating to EdgeTX on a B&W radio, you would need to use a backup saved via OpenTX Companion, and you can then load that. From EdgeTX 2.6 onwards, B&W radios also save their model and radio settings to the SD card - EEPROM is no longer used.

gagarinlg, 2023-03-18:
Can this issue be closed?

chuchotiao, 2025-02-21:
Hope you can see this, I just did the migration, I have the .otx file, but I'm unable to open it in EdgeTX Companion (v2.11.0), when trying to open a file I don't have the option to open an OTX file in the Open file Dialog

pfeerick, 2025-02-21:
You'll need an older version of EdgeTX Companion, probably latest of the
2.9 series is best as I think 2.10 had issues with some older/complex
setups. Thanks for the reminder, need to mention that in the release notes
(Companion will no longer open .otx as of 2.11)

chuchotiao, 2025-02-21:
Thanks so much, the last version that support OTX files is the 2.11, just dowloaded that and I was able to then open latest version and write the models to the radio, thanks again!!!