EdgeTX/edgetx #2179: main: SD Logging headers include initial "garbage" character for inputs

pfeerick, 2022-08-01:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

When testing #2122 I noticed that the column headers for the inputs had garbage values for the first character. This was not new behaviour introduced by this PR, it is also present in the main (a07a215). 

i.e. from some older logs...

```
Date,Time,1RSS(dB),2RSS(dB),RQly(%),RSNR(dB),ANT,RFMD,TPWR(mW),TRSS(dB),TQly(%),TSNR(dB),RxBt(V),Curr(A),Capa(mAh),Bat%(%),Ptch(rad),Roll(rad),Yaw(rad),FM,GPS,GSpd(kmh),Hdg(@),Alt(m),Sats,Rud,Ele,Thr,Ail,S1,6P,S2,EX1,EX2,EX3,EX4,LS,RS,SA,SB,SC,SD,SE,SF,SG,SH,LSW,TxBat(V)
2022-06-26,13:22:23.180,-19,0,100,12,0,6,250,-24,100,12,0.0,0.0,0,0,0.00,0.00,0.00,0,,0.0,0.00,0,0,-3,4,-1024,1,-40,-1024,-59,0,0,0,0,1024,1024,-1,-1,-1,-1,-1,1,-1,-1,0x0000000000000012,7.8
2022-06-26,13:22:23.310,-19,0,100,12,0,6,250,-24,100,12,0.0,0.0,0,0,0.00,0.00,0.00,0,,0.0,0.00,0,0,-3,2,-1024,1,-40,-1024,-59,0,0,0,0,1024,1024,-1,-1,-1,-1,-1,1,-1,-1,0x0000000000000012,7.8
2022-06-26,13:22:23.810,-20,0,100,13,0,6,250,-23,100,13,0.0,0.0,0,0,0.00,0.00,0.00,0,,0.0,0.00,0,0,-3,2,-1024,1,-40,-1024,-59,0,0,0,0,1024,1016,-1,-1,-1,-1,-1,1,-1,-1,0x0000000000000012,7.8
```

and with the nightly

```
Date,Time,1RSS(dB),2RSS(dB),RQly(%),RSNR(dB),ANT,RFMD,TPWR(mW),TRSS(dB),TQly(%),TSNR(dB),RxBt(V),Curr(A),Capa(mAh),Bat%(%),Ptch(rad),Roll(rad),Yaw(rad),FM,GPS,GSpd(kmh),Hdg(@),Alt(m),Sats,‰Rud,‰Ele,‰Thr,‰Ail,ŠS1,Š6P,ŠS2,ŠEX1,ŠEX2,ŠEX3,ŠEX4,‹LS,‹RS,SA,SB,SC,SD,SE,SF,SG,SH,LSW,CH1(us),CH2(us),CH3(us),CH4(us),CH5(us),CH6(us),CH7(us),CH8(us),CH9(us),CH10(us),CH11(us),CH12(us),CH13(us),CH14(us),CH15(us),CH16(us),CH17(us),CH18(us),CH19(us),CH20(us),CH21(us),CH22(us),CH23(us),CH24(us),CH25(us),CH26(us),CH27(us),CH28(us),CH29(us),CH30(us),CH31(us),CH32(us),TxBat(V)
2022-08-01,18:50:48.290,-45,0,100,13,0,6,10,-36,90,12,4.1,1.0,41,88,-0.03,-0.01,-1.07,0,,0.0,0.00,0,0,2,-2,-1024,1,1,-1024,-754,0,0,0,0,-808,985,-1,-1,-1,-1,-1,1,-1,-1,0x0000000000000012,1000,1500,1500,1500,2012,988,988,988,988,988,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,988,7.9
2022-08-01,18:50:48.700,-45,0,100,13,0,6,10,-36,90,12,4.1,1.0,41,88,-0.03,-0.01,-1.07,0,,0.0,0.00,0,0,0,-2,-1024,1,0,-1024,-754,0,0,0,0,-808,985,-1,-1,-1,-1,-1,1,-1,-1,0x0000000000000012,1000,1501,1500,1500,2012,988,988,988,988,988,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,988,7.9
2022-08-01,18:50:49.200,-42,0,100,14,0,6,10,-31,100,13,4.1,1.0,41,88,-0.03,-0.01,-1.07,0,,0.0,0.00,0,0,2,0,-1024,1,0,-1024,-754,0,0,0,0,-808,985,-1,-1,-1,-1,-1,1,-1,-1,0x0000000000000012,1000,1500,1500,1500,2012,988,988,988,988,988,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,988,7.9
```

specifically
```
Rud,Ele,Thr,Ail,S1,6P,S2,EX1,EX2,EX3,EX4,LS,RS
```

```
‰Rud,‰Ele,‰Thr,‰Ail,ŠS1,Š6P,ŠS2,ŠEX1,ŠEX2,ŠEX3,ŠEX4,‹LS,‹RS
```

### Expected Behavior

There not be garbage characters 😆 

### Steps To Reproduce

1. Enable logging on a model
2. Review the log file output

### Version

Other (Please specify below)

### Transmitter

Radiomaster TX16S

### Anything else?

a07a215018df0f68a6ed1edc4e6ea6103409a852

mha1, 2022-08-01:
@pfeerick I think 30e3ac33 (Basic utf8 support for B&W radios) introduced this. 

untranslated.h: The string name prefixes for sticks, pots, sliders, etc. were extended from 1 byte to 2 byte format. E.g.  for sticks:
`#define STR_CHAR_STICK     "\211"`
was changed to:  
`#define STR_CHAR_STICK     "\302\211"`

logs.cpp/writeHeader() skips the first byte to get to the raw text. 

Changing `const char * p = STR_VSRCRAW[i] + 1;` 
to `const char * p = STR_VSRCRAW[i] + 2;` should fix this.

mha1, 2022-08-01:
@pfeerick please put this PR on hold, needs testing

pfeerick, 2022-08-01:
You can mark the PR as draft (it's on the right side around where it says reviewers) if it's still a work in progress... Doesn't look like I can access it via github app on mobile.

closed 2022-08-04 by pfeerick

nightflasher, 2023-09-25:
hey there!
this garbage is back or was never removed
`CH1(us),CH2(us),CH3(us),CH4(us),CH5(us),CH6(us),CH7(us),CH8(us),CH9(us),CH10(us),CH11(us),CH12(us),CH13(us),CH14(us),CH15(us),CH16(us),CH17(us),CH18(us),CH19(us),CH20(us),CH21(us),CH22(us),CH23(us),CH24(us),CH25(us),CH26(us),CH27(us),CH28(us),CH29(us),CH30(us),CH31(us),CH32(us),`

is there a workaround? happens on both jumper t-lite (v1/v2) and heard of it and found in logs since january 2023, didn't happen before 2023 (just a coincidence?)

pfeerick, 2023-09-25:
That isn't what this issue was in relation to... which was the "garbage character" in front of each of the names of the inputs...

>`Rud,Ele,Thr,Ail,S1,6P,S2,EX1,EX2,EX3,EX4,LS,RS`
>`‰Rud,‰Ele,‰Thr,‰Ail,ŠS1,Š6P,ŠS2,ŠEX1,ŠEX2,ŠEX3,ŠEX4,‹LS,‹RS`

Your snippet looks more like what was probably an intentional change to show the outputs for each of the output channels?