EdgeTX/edgetx #2181: Charging does not stop when baterries are full

FPVian, 2022-08-02:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

The controller never stops charging the internal 18650's when plugged into USB power. Potential fire hazard.

### Expected Behavior

The batteries should stop charging when hit the upper limit of the range defined in the RADIO SETUP screen.

### Steps To Reproduce

Adjust the Batt. calib value in the HARDWARE menu.

### Version

2.7.1

### Transmitter

Jumper T-Pro

eshifri, 2022-08-02:
I'm not 100% sure about T-pro, but usually charging is HW-controlled. ETX is not involved in it.

pfeerick, 2022-08-02:
The upper and lower limits and battery calibration value are only there for the user interface (ie. battery gauge, displaying battery voltage). The radio firmware is not involved in the battery charging for any of the transmitters, but instead dedicated charge ICs or secondary processes that we don' have controll over. AFAIK the absolute nearest we get is the NV14, which actually signals if it is charging or not... but that is all. 

If the charger does not stop charging the batteries, then there is either a hardware fault or dud batteries. It would be best to measure the voltage of the batteries (with a known to to be reasonably accurate meter) while charging to ensure it is actually over-charging them (not simply staying fully charged), and then take it up with your reseller/Jumper as a warranty issue if it is.

closed 2022-08-02 by pfeerick