EdgeTX/edgetx #2182: main: widget interaction issues

pfeerick, 2022-08-03:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Regressions:
- [ ] touch: Double tap on a widget no longer fullscreens widget
- [ ] non-touch: If using rotary encoder to full-screen a built-in widget, there doesn't seem to be a way to make it exit fullscreen... except for the power button 🤪 . Lua widgets (i.e. Timer2, EventDemo) appear to be exit fullscreen properly on long RTN. 

### Expected Behavior

No comment

### Steps To Reproduce

1. Double tap on widget to your hearts content, with no fullscreen in sight...
---
1. Add a 'built-in' widget - i.e. Outputs or Models 
2. Long press ENT
3. Select said widget
4. Choose Fullscreen
5. Try to get out of fullscreen mode using long press of RTN (no cheating... no using the power button or unplugging the battery!!)

### Version

Other (Please specify below)

### Transmitter

Radiomaster TX16S

### Anything else?

For refernece, using main as of a0770beb5f8616e2f14f2828de12b3ff8b1f3397

richardclli, 2022-09-14:
I figured a problem in long scrolling list.  e.g. The protocol selection in MPM, everytime trying to scroll fast to reach the OMP protocol, nothing can be selected in the list and cannot exit the selection as well.  Will give more details when necessary.

pfeerick, 2022-09-14:
I've actually seen that sort of behaviour now that you mention it, and not in the protocol selection list, but in the main sources lists... it seems to be if you have the list too long it stops responding - I've not thought to hunt it down further so can't be sure it's not a backlight timeout issue though.

closed 2022-09-25 by pfeerick