EdgeTX/edgetx #2183: Not so much a bug as a switch test missed

MervDale, 2022-08-03:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

 In the switch test sequence , Radio Settings, Hardware, Switches.... all switches are tested but the  P 1 to 6 Butttons are not, 

### Expected Behavior

Should be included 

### Steps To Reproduce

Radio Settings, Hardware, Switches

### Version

Other (Please specify below)

### Transmitter

Radiomaster TX16S

### Anything else?

 Nightlies  I'm not sure if this has been mentioned here but I remember hearing from another user.

pfeerick, 2022-08-03:
I'm not sure if this will happen since the 1-6 buttons are actually an analog input like the pots and sliders which is treated like a set of buttons. So is tested on the analogs page (`06` for the TX16S). It would only make sense to do so if we are able to make it so you don't need to also calibrate it along with the other analogs.

MervDale, 2022-08-03:
Yes I guess you are right there, they aren't quite so obvious in the analogs showing up as only one entry but yes you do need to calibrate  them and they are not really switches which you don't have to calibrate.

gagarinlg, 2022-09-15:
It should be possible to use them without calibration. There can't be much different in the voltage the 6-POS MCU generates between the different radios.

raphaelcoeffic, 2022-09-15:
> It should be possible to use them without calibration. There can't be much different in the voltage the 6-POS MCU generates between the different radios.

This is something we should really consider. Calibrating these 6POS is really a pain.

rdeanchurch, 2023-07-17:
Not to mention that it is impossible to 'Center' an even number of switches.
Since they are discrete analog values they should not be allowed where a true or false is expected.
For instance we can use 6P-1, 6P-2.. to select Flight Modes.  
But changing the chicken feet does not change Flight Modes.
Also presently I can't find how to put any 6P, into a widget.

Either treat it like an analog or as 6 true or false values.

Well,I was going to attach .etx to show this, but .etx is not allowed

gagarinlg, 2025-01-24:
How does this beahve related to customizable switches?

pfeerick, 2025-02-03:
>  all switches are tested but the (6)P 1 to 6 Butttons are not,

While this is true ... 
> On radio with 6POS, the 6POS is both an analog source, and 6 discrete switches (6P1 to 6P6)

... when the 6POS is not calibrated, 6P1 to 6P6 do not function. Perhaps the point he was making with the 6POS "virtual switch" is that this would be another indication as to WTH is going on. And perhaps also place to put a message of "not calibrated". 

Since you've implemented some default calibrations, this will become less of an issue for new configs going forward. I think perhaps the onboarding/configuration screen needs to be improved to specifically address calibrating the 6POS when relevant, but other than that, I'm not sure it is worth the effort in adding specific testing for the 6P1-6P6 buttons. Either way, I doubt anything will change for 2.11, so will shunt this along until a final decision is made