EdgeTX/edgetx #2187: Zorro ELRS Date/time occassionly reset

pfeerick, 2022-08-05:
I have the same problem with the Radiomaster Zorro ELRS version. Date/time reset every reboot sometimes to 2000, sometimes to 1970 year. EdgeTX 2.7.1 and nightly 2.8.0 the same behavior.

_Originally posted by @Roman-Golod in https://github.com/EdgeTX/edgetx/issues/1588#issuecomment-1206107884_

Roman-Golod, 2022-08-07:
It doesn't happen, when I pull off the battery (I use the external 2s battery) without switching off Zorro. Time/date saves in that case. Maybe I can make any another experiments to help solving this problem?

nrlourenzo, 2022-11-05:
Is there any update on this? I have exactly the same issue, date resets to date 2000-1-1

Roman-Golod, 2022-11-05:
I received new quartz and capacitors from Radiomaster, replaced it, but it didn't help. Though date/time resetting now happens less often. Maybe l'll try to do it ond more time. So, i think, it's hardware related problem.

TerraMotoEU, 2022-12-03:
Got a new Zorro 4-in-1 that came with EdgeTX 2.7.1 and had the same problem: Date and time reset after reboot. Issue still persisting after update to 2.8. Mostly resets to yr 2000 but sometimes to 1970.

pfeerick, 2022-12-04:
You'll find when it resets to 1970, most likely it will be 2000 on the next boot... since 1970 is the default date of the RTC, and is from before when OTX/ETX was first written, so is then set to 2000. The problem is finding out *why* it is resetting... Companion doesn't have anything to do with setting the time, although some people seem to see it happen after a sync with Companion. I've not yet experienced this myself - probably because I never initially set the time on my Zorros, but once set they seem to have stayed set. 

This almost feels like the backup RTC battery is flat, but there must be something more to this as this would be an easily fixed occurrence (and wouldn't be evident until after several months to a couple of years of ownership)

gagarinlg, 2022-12-04:
I think someone should take a look at the RTC supply voltage line with a decent oscilloscope when switching on and off and when un-/-plugging USB.

@Risto do you have a Zorro?

Roman-Golod, 2022-12-08:
10 days ago I rebuilt the SD content and never touched the data port since then. I thought, the problem was gone, but yesterday it happened again. The clock battery is OK (3,3v).

TerraMotoEU, 2022-12-09:
Ah the battery! Mine's at 3.24v. No idea how low they're allowed to go, though.

GavoGavo, 2022-12-14:
Got mine today.

ELRS version on 2.7, and experiencing the same issue. Have noticed it go to both 1970 and 2020.

I have mostly been using it plugged in to the PC for flying Sim. 

I wonder if the process of plugging in and out is related to the cause?

It's very frustrating.

TerraMotoEU, 2022-12-14:
Contemplating whether I'd send it back to The Netherlands for replacement or not. After ordering from Radiomaster EU, PostNL took forever to even move my order. If I was sure it's something that can be fixed via fw update, I'd hold on to it and wait.

Aside from messing up file names (logs, model backups...) is there any other risk coming along with this issue, that we should be aware of, or is that all there is?

GavoGavo, 2022-12-15:
Were these all new radios?

Did anyone else notice sticker residue on the cathode/positive side of the batteries, when they removed the clear-and-red battery tab that breaks the circuit in transit?

I noticed when I removed mine that quite a lot of residue was left over. I used the tab to remove the remaining residue but I tried charging the batteries in my XTAR charger and noticed they were not reliably making a circuit. I tried getting more residue off and they seemed kind of OK after that. 

I tried cleaning them futher just now.. some residue did seem to come off and the charger now has no problem detecting them. I wonder if there might be a physical component here... poor contact perhaps. I don't think so because I would think that any drop in contact would switch the device off but perhaps it's worth a thought?

If you do try and clean the terminals, be really, really careful and read up about how to do it before you do. These batteries can be dangerous and though they're protected, a short could be catastrophic. 

I do note that I have been unable to repeat the problem so far today anyway, and also upgraded firmware to 3.8.

I am going to keep trying to reproduce the problem. If it happens again I will lodge a warranty request with Radiomaster to see what they say. Hopefully this is just a small batch of units. I do wonder if we all got devices from the same batch maybe.

andreo32, 2022-12-15:
I got an answer, when I addressed the same question to radiomaster support:
`This is a known issue, a STM related bug that we've communicated to edge TX, who should be addressing the issue with a future update.`

andreo32, 2023-01-01:
Hi!
There's a guy in [RC groups who explaining RTC reset](https://www.rcgroups.com/forums/showpost.php?p=50318429&postcount=1210) issue by replacing original RTC battery to new - a little bigger that original.
Maybe is just a to weak battery problem...
Br,

TheIsotopes, 2023-01-07:
@pfeerick Is there anything news on this topic?

I'm having the same problem and it's frustrating.

Dopossum, 2023-02-09:
I had the same problem, Time and Date resetting to 2000 or 1970.
 I have the Radiomaster Zorro Elrs V2.0  running under EdgeTX 2.7. 
In the menu SYS/Radio Setup , where also Date and Time are set 
there is a Checkbox option " Adjust RTC" . I enabled it.
I have rebooted several times since and Date and Time where everytime correct.

raphaelcoeffic, 2023-02-09:
“Adjust RTC only make sense when a GPS is connected that is then used to fetch the current time. Your issue is most probably related to the RTC itself, and it’s supply power. I believe we have fixed such an issue that related to some brownout level. At best you should update to 2.8, which has the fix included.

KC748, 2023-02-15:
Hi everyone. I'm the guy who replaced the RTC battery with a bigger one mentioned above by andreo32.

My Zorro is on v2.8.0 and just as pfeerick has described above ( and in #1588), the RTC will reset to 1970 if the Power button is pressed while the main battery is removed. And then reset to 2000 on the next reboot. The RTC retains it's setting in subsequent boot cycles after that.

It doesn't seem like this issue is fixed yet?

> You'll find when it resets to 1970, most likely it will be 2000 on the next boot... since 1970 is the default date of the RTC, and is from before when OTX/ETX was first written, so is then set to 2000. ...
> 
> This almost feels like the backup RTC battery is flat, but there must be something more to this as this would be an easily fixed occurrence (and wouldn't be evident until after several months to a couple of years of ownership)

kf4mot, 2023-02-15:
> It doesn't seem like this issue is fixed yet?

It's most definitely not fixed.

chemist132, 2023-03-15:
Same here. i had mine for 1.5 years without trouble and now the date and time are buggy(: No solution yet.

mixMugz, 2023-03-15:
Any news about bug? Zorro 2.8.1 EdgeTX still reset date/time after reboot(poweroff)

KC748, 2023-03-16:
Mine doesn't reset after a reboot. It only resets at certain instances as described above.
If it is resetting after a reboot, the coin cell may be going flat. The cr1220 are small and only last about 1.5-2 years.

I don't know if this is related, but after replacing the cr1220 with a brand new Duracell cr2032, the beginning 3.33v has dropped to 3.08v in 2.5 months. This drop seems excessive. 

[image]

[image]

mixMugz, 2023-03-16:
3.08V is not enough?

pfeerick, 2023-03-16:
I would think so, it just seems unusually quick for it to drop from 3.33v to 3.08v - although 3v is the nominal voltage. 

It should be a lithium primary cell (thus non-rechargable).

mixMugz, 2023-03-16:
So. Another test.
Bought the new CR1220. Now voltage show 3.3 volts.
Seting date/time.
Step one - power off button. Date still correct after power on.
Step two - power off by removing battery. Date still correct after power on.
Step three - connect wia usb (joystik hid) then power off. Date reset after power on.

Now every time i power off radio - date will be reset. Sometimes date will keep, but reset after second/third reboot

KC748, 2023-03-17:
While doing some searches in OpenTx for RTC issues, I came across this one from way back in 2015. It looks like this RTC reset issue has been around for quite a while and on various radios.
https://github.com/opentx/opentx/issues/2626 
In this issue thread, Mike Blandford has a possible code fix, 
https://github.com/opentx/opentx/issues/2626#issuecomment-128933531
and there is also a hardware fix (for the Taranis X9E) by replacing a capacitor with a larger value capacitor. 

Any chance of implementing either of these solutions for the Zorro?

pfeerick, 2023-03-17:
The comment that closed that issue is rather telling also... 
> It's an hardware problem, please contact FrSky (the 100uF is not enough, they will use a 470uF instead)

The code that Mike Blandford commented on is already in OTX and ETX, and has been for about four years - was added as part of https://github.com/opentx/opentx/pull/6501 - and is used by basically all the B&W radios that have a soft power button (rather than switch). 

So it seems for that particular radio, it was a hardware issue through and through, and was also very intermittent, so really does fit the pattern we're seeing here. So the question now is probably going to be how to confirm it is an issue related to a capacitor value being too low (and which one)?

Nice bit of investigation... could be getting closer! :)

AlxSlk686, 2023-03-19:
Hello! I'm having the same problem with my zorro radio which is using firmware 2.8.1 I found a clear dependency on resetting the date and time. If you turn off the radio immediately after exiting the system menu or from it, the date and time are reset, if you exit the menu and wait a few seconds, the date remains unchanged.

KC748, 2023-03-20:
> The comment that closed that issue is rather telling also...
> 
> > It's an hardware problem, please contact FrSky (the 100uF is not enough, they will use a 470uF instead)
> 
> The code that Mike Blandford commented on is already in OTX and ETX, and has been for about four years - was added as part of [opentx/opentx#6501](https://github.com/opentx/opentx/pull/6501) - and is used by basically all the B&W radios that have a soft power button (rather than switch).
> 
> So it seems for that particular radio, it was a hardware issue through and through, and was also very intermittent, so really does fit the pattern we're seeing here. So the question now is probably going to be how to confirm it is an issue related to a capacitor value being too low (and which one)?
> 
> Nice bit of investigation... could be getting closer! :)

Thanks for the additional information.

I've reread #2626, and according to projectkk2glider, in his post, https://github.com/opentx/opentx/issues/2626#issuecomment-128792600,  the additional 470 uF capacitor is added in parallel to Vcc (3.3v) and GND.

>add additional capacitor in parallel somewhere between Vcc and GND lines,
>
>That trace is your Vcc line and should have 3.3V when the radio is powered on. The GND is on the copper fill surrounding it

I've located the 3.3v regulator (A1117B 3.3) on my Zorro and did some continuity tests and came up with the diagram below. Conveniently, there is a 3.3v and GND pads close to the regulator. 

Following projectkk2glider's instructions, I'm about to add a 470 uF capacitor on those pads, Does anyone see any harm in doing that?

[image]

KC748, 2023-03-21:
Ok. I gave it a day for someone to talk me out of it...  Crickets. So here it is. :)

Following projectkk2glider's instructions, I added a 470 uF 10v capacitor (recommanded is 6v or higher) on those pads (3.3v and GND) and tacked it down with a bit of hot glue over a piece of electrical tape for insulation. 

On my Zorro (v2.2), a sure way to cause the RTC to reset is to press the Power button when the main batteries are out. The Power button would briefly flash blue. The RTC would reset to 1970, then to 2000 on the subsequent reboot.

So far, I haven't been able to cause the RTC to reset after adding the 470 uF capacitor. I'm quite satisfied with the result.

Tested:
* Power button press after removing the main battery, briefly as well as holding for more that 5 minutes.
* Flash firmware (v2.8.1) with USB connected and via SD card in bootloader mode. (Power On with horizontal trim to center)
* Flash firmware via USB with power Off.
* USB connected in Joystick mode and Storage.

Even after bricking the radio a couple of time, by pulling the USB before ejecting the Zorro's drive from the computer, the RTC did not reset after unbricking.

Disclaimer: 
Do the mod at your own risk. I'm not an expert. :)
If anyone tries this mod, some feedback would be nice.

[In the pics below, note the offset placement of the capacitor, to clear the gimbal sensor housing and wires.]

[image]
[image]

mixMugz, 2023-03-21:
Good news! I need to try with this stuff
[image]

KC748, 2023-03-21:
It's curious that the tantalum capacitor above the 3.3v regulator is a 47uF capacitor (476J) and not a 470uF (477j). I'm wondering if Radiomaster screwed up and put the wrong value there.

pfeerick, 2023-03-21:
> On my Zorro (v2.2), a sure way to cause the RTC to reset is to press the Power button when the main batteries are out. The Power button would briefly flash blue. The RTC would reset to 1970, then to 2000 on the subsequent reboot.

Oh, thanks for that - I think I have had this happen myself then... so will try to reproduce this, and then try the mod also if it happens repeatably for me. I'm now also curious as to whether the ELRS zorro is more prone to this than the MPM one, as it *could* be a little more power-hungry at startup and be causing a brownout, hence why the bigger cap needed. Anyway, something to pass on the RM for them to investigate also.

AlxSlk686, 2023-03-22:
I installed back the firmware from RM 2.7.0 and all the problems with resetting the time finally disappeared. No matter how I try to reproduce this problem. All problems start on firmware 2.7.1 So it still looks like this is a software problem and not hardware

richardclli, 2023-03-22:
It is interesting tough change between 2.7.0 and 2.7.1 are mainly UI related.  Cannot spot any change that may affect the RTC.

pfeerick, 2023-03-22:
Indeed... IIRC the brownout PR was part of 2.7.0, so it won't be that. and rolling both FW and bootloarder back to 2.7.0 don't make any difference to either my MPM or ELRS Zorros for the "power on when batteries removed" triggering of the RTC being reset.

AlxSlk686, 2023-03-22:
> Интересно, что изменения между 2.7.0 и 2.7.1 в основном связаны с пользовательским интерфейсом. Не удается обнаружить какие-либо изменения, которые могут повлиять на RTC.

I myself cannot understand why this is so. My zorro was originally 2.7.0, then I started updating it and there was a problem. I reverted to 2.7.0 Now it's been another day since my last post and the time hasn't reset as usual. I periodically try to cause the problem with different actions but everything is in order. Mystic))))

TheIsotopes, 2023-04-01:
The capacitor mod helped me ... since @KC748 posted it i have it in action and i've seen no clock reset.

KC748, 2023-04-02:
That’s good news! Could I ask which version of Zorro you have? 

There are couple of reports on RCG, both ELRS versions, that rests at every boot. It would be nice to give them some hope. Mine is the 4in1 and  it has been rock solid since adding the cap.

TheIsotopes, 2023-04-02:
>  Could I ask which version of Zorro you have?

4in1,  Zoro Board v1.9

Roman-Golod, 2023-04-03:
Great work, thanks! I've just added 1000 uF capacitor and no resets after some on/off cycles! Zorro ELRS board 2.2

mixMugz, 2023-04-09:
So. I can confirm. The mod is working.
Putting 470uF / 6.3V on board and I no longer have any case with time/date reset.! 

[image]
[image]

bbolgradov, 2023-04-22:
Excellent, thanks! Mod works on Zorro Board v2.5 with ELRS.

dominikm23, 2023-04-27:
There is a socket on the other side of the board which is directly connected to the 3.3V trace. I just connected a matching plug to a 470uF capacitor and plugged it in. No need to solder or disassemble the circuit board.

I have not had a single date reset since this change and it used to do it pretty much every time I turned it on.

[image]

Berlinologe86, 2023-05-06:
Raeiomaster says it is a bug on edgetx? But why is it only on This Radio 😔
[image]

can a edgetx dev confirm This ?

chemist132, 2023-05-06:
I have a Tx16S with edge and the same problemen. I orderbedrag a 470 muF condensator.

Shpek, 2023-05-09:
I have a BETAFPV Lite Radio 3 Pro 2.8.2 with exactly the same problem - date is set to 01-01-1970 on every reset. RTC battery is 3.1V. I also tried with the battery from my TX16S (which is working fine) - no effect. Tried the cap mod with a 470uF electrolytic and 100uF tantalum - still resetting. Can't downgrade to 2.7.0 since it doesn't have support for Lite Radio 3 Pro.

Shpek, 2023-05-09:
#

> > Tried the cap mod with a 470uF electrolytic and 100uF tantalum - still resetting.
> 
> I tried 470uF with no effect, but 1000uF helped me. Try to increase capacity.

1000uF made it reset to 01-01-2000. Tried with 4700uF - same thing.

pfeerick, 2023-05-10:
> Radiomaster says it is a bug on edgetx? 
> can a edgetx dev confirm This ?

That's interesting... to the best of my knowledge is it related to a design choice made by RM to ensure you don't get any mid-air power-offs due to oxidisation of the battery contacts. There are two electrolytic capacitors that are holding enough charge to cause the RTC circuit to stop working properly, hence the time/date reset. Adding more capacitance is probably allowing the MCU to start enough to realise it's in a brownout condition and stop itself properly. Removing them would also have the same effect, but then you'd want to make sure you take care of the battery contacts to prevent intermittent power. 

Other than that, this is a nearly impossible issue to fault-find, as outside of the Zorro and an unusual/extreme sequence of events, there does not seem to be any pattern that allows us to reliably reproduce the fault on our end.

Thomas-Cee, 2023-05-12:
So @KC748 , could I remove the surface mount 47μF and remount a larger surface mount cap like @mixMugz pointed to? Does it need to be an electrolytic? 

[image]

KC748, 2023-05-13:
I'm really not the right person to ask, to be honest.
My sense is that the tantalum would be ok instead of an electrolytic. But Personally, I'd just add it to the 3.3v/GND pads. It's way less risky. 
Hopefully someone will chime in with a more definitive answer. Or otherwise, you're welcome to blazing the trail on this one. :)

zatalian, 2023-05-13:
I noticed that if I keep pressing the power off button longer on turn off, time is not reset. After the 4 dots the sleeping face appears and stays on as long as i keep pressing the power button. If I do that for a few seconds, the clock is not reset.

This made me wonder if maybe the haptic feedback between the dots and the sleeping face is causing the reset. I turned haptic feedback off and for now. No resets since...

kf4mot, 2023-07-06:
I replaced the pathetic little coin cell in my Zorro 4 months ago. Now it's down to 3 volts again and I got another RTC reset.
Now I soldered in a 2032 size cell. Time will tell if it stays set any longer, but with roughly 8X capacity hopefully it last a couple years.

MAcevedo03, 2023-07-19:
Below is a reply I received from Marketing at Radiometer after I wrote to them about my resetting issue with the 4 in 1 RC.
It's sounds like you can send the RC back, however I'm not sure the "new" RC won't have the same issue.

Thanks for your contact. On the SYS-Hardware page of the remote control, you can see the RTC Battery voltage. When it is lower than 2V, the calendar and time may not be saved correctly, indicating that it needs to be replaced. You can also tick the Check RTC on this page. When the button battery is out of power, the remote control will pop up a prompt message.

lightbulbjim, 2023-07-19:
Since adding a 470uF capacitor across the 3.3V rail a few months ago I haven't had any more resets. Not sure whether to put the fault down to poor design or cheap (out of spec) components.

CountParadox, 2023-07-21:
I just received my transparent zorro, I am experiencing the RTC reset issue also, I will try the cap mod and see if it helps. Just wanted to let people know its still an issue on new units :)

Thomas-Cee, 2023-07-21:
I wonder how a larger cap install, and how a larger 2032 button cell that @kf4mot referenced interface with eachother? 
Interesting that they both seem to cure the reset issue.
Does the cap, provide a brief burst of current instead of being drawn from the battery, causing the battery to last longer?
That's interesting. I do wonder what at a circuit level is happening in there...

kf4mot, 2023-07-22:
Without schematics we're all just kind of speculating. I've added caps and a bigger RTC battery because I'm a belt and suspenders sort of guy. I don't see that either _conclusively_ solves the problem though unfortunately.
I've also considered adding a ceramic cap parallel with the RTC battery since coin cells can't handle high current draw. If the RTC ever needs a higher current pulse that could help. I have no evidence that's a issue though either. The larger CR2032 would handle higher current pulses better than the stock RTC battery on it's own by being larger.

TrevorCrouch, 2023-08-07:
I've had quite a few Frsky, TBS RX which have worked fine. Splashed out on the Zorro Max a few weeks ago and have the rest issue.  It's annoying for sure and the above remedies appear to 'possibly work for a while'. My question is, does it effect anything if it does reset back to yr 2k or 1970??

TrevorCrouch, 2023-08-07:
> It will affect logging and backups. Maybe the last modified time of files if you use the SD sync feature in Companion.
> 
> Adding the capacitor has fixed it reliably on my two Zorros.

Thanks LBJ, did you install just the 470uf?

TrevorCrouch, 2023-08-07:
Can you confirm the dimensions of the 470uF 10v you used please Jim

[43 more replies]

closed 2025-01-24 by gagarinlg