EdgeTX/edgetx #2190: S2 not working properly on RM T16s MK II in v2.7.1

olli-e, 2022-08-05:
When installing 2.7.1 on my RM T16S Mk II, S2 is only using half the way - it can either be -100 to 0 or 0 to +100, but not the full range as S1 does. I calibrated and did everything possible after blank install and re-tried three times, but it is always the same - S2 is not fully usable. Fell back to v2.7.0 and everything is working again. Just as an Info - will be using v2.7.0 until fix.

Thanks for all your hard work fellows!

olli-e, 2022-08-05:
Thanks, I'll try the special formatting - I had the 32 Gig SD Card, not the stock one, formatted in FAT32, but will try from a blank system and get back to you, once done.

pfeerick, 2022-08-05:
Ok, in that case, we are also tracking an issue with radiomaster where larger SD cards play up... so you could also be suffering from that...

olli-e, 2022-08-05:
I will use the 32 Gig SD and will also try the 256MB card that is delivered with it and write back about the results

olli-e, 2022-08-05:
Formatting a 32Gig SD Card with 4K Cluster is not working btw - you can select it, but get an error message - default is 8K-64K
[image]
w

When I use a partition size of 4GB, I can successfully use 4k sector size - so I'll try that. Works also for the original 256MB SD card .

Will get back with more info.

pfeerick, 2022-08-05:
The sector size is more to do with audio - seems to prevent distortion/skipping while reading the files... but other than that it shouldn't (famous last words) affect anything. It will be interesting to see if the 256MB card is fine, and the 32GB plays up...

olli-e, 2022-08-05:
First trying all from new - fresh download of all items and usage of SanDisk Extreme 32GB

**Step 1:**
SD Card 32Gig, 4 GB partition size (Fat32/4096)
Flashing with edgeTX flasher runs without problem.
SD Card is not recognized by flasher, whether it is in the Tx or in the PC (administrative permissions don't make a difference)
after coping the SD Card content manually to SD Card and restarting of Tx:
Startup results in "Storage Warning - Bad radio data"
Didn't do anything further and moved to step 2.

**Step 2:**
SD Card 32Gig, 4 GB partition size (Fat32/4096) - fresh formatted
Flashing with bootloader (2.7.1) runs without problem.
SD Card content is still on the SD Card - did not do anything with it
Startup results in "Storage Warning - Bad radio data"
After clicking on ENT, next step in TX is Storage Preperation.
The Tx then moves into Calibration mode and all is clean
At this point all but S2 are working correctly (S2 0 to +100) and can be calibrated.

I stop it now, switch off the Tx and start it again - now I don't get the bad radio data error message - and the calibration does not start again - S2 is still not working. The SD Card still has the same configuration "FAT32 INIT13" as before.

**Step 3:**
SD Card 32Gig, 4 GB partition size (Fat32/4096) - fresh formatted
Checked the SD Card - is still the same filesystem, no changes made. Formatted it new 4096MB, 4k sector size. Copied the blank SD card content and firmware on it and flashed it using the bootloader.
Flashing went well - exit and boot into EdgeTX BTW S2 is centered
Again "Storage warning - Bad radio data"
Same behavior as above - S2 only using half the way and calibration does not change anything.

**Step 4:**
SD Card 32Gig, 4 GB partition size (Fat32/4096) - fresh formatted
Checked the SD Card - is still the same filesystem, no changes made. Formatted it new 4096MB, 4k sector size. Copied the blank SD card content and firmware (nightly this time) on it and flashed it using the bootloader.
Flashing went well - exit and boot into EdgeTX BTW S2 is centered
**No Bad Data warning, BUT this time

it is NOT S2 that has the problem, but LS - S2 is working fine, LS is using -100 to 0 only, regardless of calibration - and yes it was centered when I entered calibration.**

Will now test everything the same way with the 256 MB SD Card that came with the Tx.

olli-e, 2022-08-05:
First trying all from new - fresh download of all items and usage of Noname Original SD 256MB - formatted in Fat32

Step 5:
SD Card 256MB (Fat32/4096)
Flashing with edgeTX flasher runs without problem. **This time I made sure to eject the STM 32 Loader**
SD Card is not recognized by flasher, whether it is in the Tx or in the PC (again - administrative permissions don't make a difference)
after copying the SD Card content manually to SD Card (inserted in Tx and restarting of Tx:
Startup results in normale startup - **perhaps the eject of the STM Loader made the difference.**
Now All sliders, potis and buttons work correctly.

Step 6:
Retrying Step 1 with the eject of the STM Loader and the 32GB SanDisk Extreme - Partition 32GB Fat32
Flashing with edgeTX flasher runs without problem. This time I made sure to **eject the STM 32 Loader**
SD Card is not recognized by flasher, still - copying manually while inserted in Tx.
**Eject SD Card and exit bootloader**
### Now everything is running well - no problems after calibrating.

So, as a summary, it obvioiusly depends on if you hit the eject button everytime before you dicsonnect the Tx or SD-Cards, otherwise the data gets corrupted somehow and this ends in a weird behavior.

**No problems on your side afte rall!**

Thanks
Olli

gagarinlg, 2022-08-05:
Are you sure that your card reader is configured as a removable driver in your OS?
Sounds like the buffering is enabled

olli-e, 2022-08-05:
It didn't matter, if I had the SD Card in the Tx or in the PC, so if both are configured wrongly, then this might have that effect, but honestly I have no idea how to check for that. Can you give me a hint?

closed 2022-08-05 by olli-e

gagarinlg, 2022-08-05:
https://docs.microsoft.com/en-us/troubleshoot/windows-server/backup-and-storage/turn-disk-write-caching-on-off

You can try this on the SD card device

pfeerick, 2022-08-06:
> So, as a summary, it obvioiusly depends on if you hit the eject button everytime before you dicsonnect the Tx or SD-Cards, otherwise the data gets corrupted somehow and this ends in a weird behavior. Even when you use the flasher - eject before disconnect!

Turning write caching off if for some reason it is on is a good first step, but generally it's good to eject regardless of whether it's USB flash drive, microSD, etc... just gives that peace of mind that everything has finished writing, and minimises chances of stuff going wrong... 

Having said that, I'm glad it is working for you now... fingers crossed this is indeed all it was. Regarding flasher issue not recognising , I think I encountered this before, and it's a bug in Flasher, so that is a different issue that needs to be identified and fixed.