EdgeTX/edgetx #2191: Tx16s loosing default settings after three or four restarts

davx35, 2022-08-06:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Following three or four restarts of tx the radio reverts back to default settings and has to be re calibrated. This did not happen under open tx.

### Expected Behavior

Retain its memory following restarts.

### Steps To Reproduce

Turn on tx. Then power down.

### Version

2.7.1

### Transmitter

Radiomaster TX16S

Eldenroot, 2022-08-06:
Cannot reproduce...

gagarinlg, 2022-08-06:
This is probably linken to #1481

pfeerick, 2022-08-07:
Also, RM have commented that they have 

> found there is a chance that the SD card initialization will fail on boot, in such a case the system may be stuck for 20s or more.
...
> reduces the problem of SD card data corruption requiring the card to be formatted and rebuilt.

Ben said it seems to be more prone to happen as the SD card size increases... so it seems like something else to look into...

mha1, 2022-08-07:
> Also, RM have commented that they have
>
>
> Ben said it seems to be more prone to happen as the SD card size increases... so it seems like something else to look into...

I had this SD stuck issue at boot. Occurred randomly and when it struck it took up to 40s to boot. I solved this by getting rid of the Samsung EVO SD Card and replacing it with a Sandisk. Never had this problem again. Just a few days ago I met a guy with the same problem.  Guess which card he had? There is definitely a problem with certain SD Cards. If you want to chase this get a hold of a Samsung EVO SD Card 32GB or 64GB.

pfeerick, 2022-08-08:
Thanks for that... that seems to be what I've seen also... my lockups seemed to be with 32GB EVOs... (and never when I was in a position to run the debugger 😆 ) so I've now got a bunch of Sandisk Extreme 32GBs as well to see how that goes. I would normally have used 16GB EVOs as I've found them generally very reliable and consistent but just can't buy them any more :(

mha1, 2022-08-08:
@pfeerick I'll try to acquire the guy's card I talked to recently. This card definitely shows the problem. If I succeed I'll send it to you.

pfeerick, 2022-08-08:
Don't try too hard just yet... we're chasing a few squirrels with RM atm as
they have some reproducible failure cases and are set up to do some cycle
testing on multiple radios... ;)

gagarinlg, 2022-09-09:
@davx35 can you please retest with a current nightly?

davx35, 2022-09-12:
> 

Have installed latest nightly. Looking very positive at the moment, time will tell. Will carry on testing over week and post back at weekend. Think there may be another bug though. I tried to flash latest multi protocol module and it failed, had to re flash open tx for it to work.

pfeerick, 2022-09-12:
Fingers crossed. And there was a mention of a issue with MPM recently, so thanks for the reminder to look into that.

davx35, 2022-09-16:
> Fingers crossed. And there was a mention of an issue with MPM recently, so thanks for the reminder to look into that.

Have been testing all week, everything seem fine other than ghost lua is now not working. Like new interface by the way.

gagarinlg, 2022-09-22:
> Have been testing all week, everything seem fine other than ghost lua is now not working. Like new interface by the way.
> […](#)

Thank you very much for testing.

closed 2022-09-22 by gagarinlg

pfeerick, 2022-09-22:
>  other than ghost lua is now not working. 

You use Ghost? :) What's wrong with the Lua? 

Would you mind giving this version of the widget a spin and seeing what you think... I started making some changes to it but got distracted with other stuff... be interesting to see if it works for you or could still be improved. 

[ghost-widget.zip](https://github.com/EdgeTX/edgetx/files/9625312/ghost-widget.zip)