EdgeTX/edgetx #2194: Add acoustic feedback for SYS/RTN/PAGE/TELE and MDL hard keys

mha1, 2022-08-06:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

No problem and seems to be as is implemented in 2.7.1.

### Describe the solution you'd like

For consistency I propose to add acoustic/haptic feedback just like for the roller enter key, depending on radio setup Sound/Haptic settings "All".

Eldenroot, 2022-08-06:
Good idea, why not

mha1, 2022-09-14:
clesed with https://github.com/EdgeTX/edgetx/pull/2201 and https://github.com/EdgeTX/edgetx/pull/2193

closed 2022-09-14 by mha1