EdgeTX/edgetx #2195: Standalone Simulator W10/64Bit (nightly 62692dba August 4th)) - no simultated touch with mouse click

mha1, 2022-08-06:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Companion

### Current Behavior

Can't simulate "touch" with mouse clicks in standalone simulator (2.8.0 - main nightly 62692dba dated August 4th). 

Works fine if Companion built in simulator is used.

### Expected Behavior

Mouse clicks are accepted as touch events in standalone simulator

### Steps To Reproduce

Run simulator.exe (2.8.0 - main nightly 62692dba dated August 4th)
Position mouse pointer within screen
Click mouse
No action

### Version

Other (Please specify below)

### Transmitter

Radiomaster TX16S

mha1, 2022-09-12:
anyone?

mha1, 2022-09-14:
@pfeerick this is still not solved and very annoying. And even more annoying with the to rotary wheel clicks per mouse wheel increment. 

Any idea where to look for to get at this? I am by no means a Windows expert but willing to put in some effort. Can you say if Linux and Mac standalone simulators are good?

gagarinlg, 2022-09-14:
Try setting the mouse scroll spreed in windows to one.
If that helps, somewhere in the simulator, this has to be done for the simulator window.

mha1, 2022-09-14:
@gagarinlg 

Hi, there is two separate mouse issues:
- https://github.com/EdgeTX/edgetx/issues/2188 which is valid for standalone simulator and companion built in simulator where one increment of the mouse wheel leads to jumping to next but one menu (like two rotary wheel increments). 
- and this one which describes the problem where no mouse button click is accepted when clicking into the simulated screen area (no touch event possibel). This works in the companion built in simulator bit not in the stand alone simulator

I believe you are referring to the first issue but unfortunately changing any of the mouse button or wheel parameters in Windows's control panel makes doesn't make any difference to both of the problems.

Who knows this part of Companion/Simulator best?

gagarinlg, 2022-09-14:
Ah sorry, I was on my phone and missed that this was a different issue.

mha1, 2022-09-14:
No problem, thanks for your suggestion. I'd say there was some kind of regression around handling mouse input in Companion/Simulator which needs to be resolved. I am looking for initial pointers into the area. Where to start?

gagarinlg, 2022-09-14:
There is this QT slot in targets/simu/opentxsimulator.h
rotaryEncoderEvent

Which is connected in 
companion/src/simulation/simulateduiwidget.cpp

I would start there

pfeerick, 2022-09-15:
> Can you say if Linux and Mac standalone simulators are good?

A WSL2/WSLg build shoed the same mouse issue as #2188, but I haven't booted into native linux to do proper A/B test.

I had a look at this with a build of simulator from a week ago - https://github.com/EdgeTX/edgetx/commit/246b061a - and mouse as touch seems to be working? Just make sure you don't click on any widgets, otherwise the touch event goes to them, not the system. i.e. the top bar logo is always a safe place test to test.

mha1, 2022-09-15:
I completely wiped the 2.8 Companion install and reinstalled again. Seems to work now. Maybe Windows didn't like many many reinstalls on top of each ...

closed 2022-09-15 by mha1

pfeerick, 2022-09-16:
lol... it could be, but its more picking the wrong monkey from the barrel that gets me 🤪

mha1, 2022-09-16:
This should be useful: https://www.youtube.com/watch?v=SHNEWlDiXEI