EdgeTX/edgetx #2200: For edgetx built-in crsf protocol add use

xiaobaishangche, 2022-08-10:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

The popular use of elrs now makes the built-in RF modules of some remote controls deprecated. Most of the built-in interfaces can be DIYed into built-in elrs social-level modules. I hope EdgeTX developers can add the built-in crsf protocol, thank you.

### Describe the solution you'd like

Turning on the built-in crsf protocol makes the player DIY still the built-in tuner of his own elrs

pfeerick, 2022-08-10:
I'm not quite sure what you are asking for here. If you mean can EdgeTX make it so the builtin RF be used as ELRS, then that is well beyond the scope of the EdgeTX project. EdgeTX is not responsible for the RF link, that is the responsibility of the built in RF module.

If you mean for an internal module to be able to be replaced with an ELRS module, then assuming it is wired up correctly, EdgeTX already supports this.

raphaelcoeffic, 2022-08-10:
> If you mean for an internal module to be able to be replaced with an ELRS module, then assuming it is wired up correctly, EdgeTX already supports this.

If I may add:
- only full duplex ELRS modules (so either from Radiomaster, Jumper, or DIY),
- only on radios that have a full duplex UART on the internal module interface (ex; legacy radios from FrSky don't have this).

Johnex, 2022-08-22:
And to help out even more, maybe he just didn't find the option for it.

In 2.7 the setting for internal module type was moved. Press SYS button, go to Hardware tab, scroll down to Internal RF. You can change there to internal CRSF, it will let you use Crossfire / ELRS modules internally in the radio, either via official modules (Radiomaster ELRS module for example) or hacked in (you can put a Crossfire Nano TX inside the radio and just wire it in).

caixiaohai0, 2022-08-24:
> If I may add:
>
> * only full duplex ELRS modules (so either from Radiomaster, Jumper, or DIY),
> * only on radios that have a full duplex UART on the internal module interface (ex; legacy radios from [...]

For the old FRSKY transmitter, is there any way to support built-in CRSF or multi-protocol, can we support full-duplex URAT function in the vacant IO port of MCU?

raphaelcoeffic, 2022-08-24:
@caixiaohai0 theoretically, this is just about defining the proper pins/UART in `hal.h`. One could possibly use the debug UART on some targets.