EdgeTX/edgetx #2202: TX12 MKII switch warning labels just show "S" instead of switch letter

rburrow87, 2022-08-12:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Switch warning settings on the transmitter just show "S" for all the switches.

### Expected Behavior

Should show B/C/E/F instead of "S".

### Steps To Reproduce

Go into model settings and look at the switch warning setup.

### Version

Nightly (Please give date/commit below)

### Transmitter

RadioMaster TX12 / TX12MK2

### Anything else?

Or maybe this is pulling it up in snake language instead of English?

Nightly c41de59

[image]

pfeerick, 2022-08-12:
Well, that's just being fussy :P What's wrong with them being 'S' and you having to guess which is which? 😆

pfeerick, 2022-08-12:
That seems to have fixed it... looks like it was an overlooked LVGL/UTF-8 change... there's an extra character there now that mucks things up every now and then.

commit b74cd109 (2022-08-12): fix: Showing wrong letter for switch warnings
Resolves #2202
closed 2022-08-12 by pfeerick