EdgeTX/edgetx #2203: MPM not showing RF fine tune value

rburrow87, 2022-08-12:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

When setting the `RF Freq. fine tune`, it becomes blank when you leave the RF menu and return.

### Expected Behavior

Frequency fine tune value should be displaying in the field.

### Steps To Reproduce

Go to model settings using either an internal or external MPM, choose a protocol that has fine tuning, and attempt to set it. If you back out of the menu and go back in, it becomes blank again.

### Version

Nightly (Please give date/commit below)

### Transmitter

Radiomaster TX16S

### Anything else?

This is present with nightly c41de59. Works fine on the Zorro and TX12 MK2.

It seems to be saving the value, it's just not displaying it when the RF menu loads. If I set it to 1 and back out, go back in, and use the touch screen to increment it by 1, it goes to 2. Then if you back out and do it again, it goes to 3, etc.

I know, I know. Very fussy today. 😬

pfeerick, 2022-08-12:
Fussy indeed... wanting stuff to work 😆 ... nice catch! :

pfeerick, 2022-08-12:
This seems to be a problem whenever the NumberEdit box is shown there - i.e. FlSky2A has a servo update rate input box, and the current value if it isn't being loaded either. So I'm thinking there is something wrong here => `/radio/src/gui/colorlcd/mpm_settings.cpp`

eshifri, 2022-08-12:
I do not see STR_MULTI_RFTUNE in the mpm_settings.cpp. And STR_MULTI_SERVOFREQ  is a choice. Am I looking at the wrong file?
https://github.com/EdgeTX/edgetx/blob/b74cd10927f20f919a2dcd697732e0b01cf45c2f/radio/src/gui/colorlcd/mpm_settings.cpp#L179

pfeerick, 2022-08-12:
I beive it is loaded dynamically by parsing the responses of the module, which is why it's hard coded. And the fact the Servo refresh is a choice is confusing since it actually comes up as a number edit field (although I didn't check DSM, maybe it actually does match for that proto).

eshifri, 2022-08-12:
I thought servo refresh was either 50 Hz for analog or 500 Hz for digital servos.

pfeerick, 2022-08-13:
Note I said "I think the other is `STR_MULTI_SERVOFREQ`" ... not that it *is* that string... ;)

And no, on DSM it is a choice, and is either 22ms or 11ms.

[image]

It looks like for FlSky2A it should be a number edit, and should be accepting a number between 0 and 70

>  Servo refresh rate. A value of 0 gives 50Hz (min), 70 gives 400Hz (max). 
https://www.multi-module.org/using-the-module/protocol-details/flysky-afhds2a

eshifri, 2022-08-13:
My bad: forgot about DSM.
In the native FlySky (NV14) version it is 50 and 500 (or 400).

pfeerick, 2022-08-13:
Ok, brain is functioning correctly now 😆 

```
  sr_line = new MPMServoRate(this, &grid, moduleIdx);
  ab_line = new MPMAutobind(this, &grid, moduleIdx);
```

Are only used for MODULE_SUBTYPE_MULTI_DSM2, which explains the servo choice in that instance. 

When the proto is FlSky2A, the edit field is turned on, and the `(refreshrate−50)÷5​` formula is being applied so that the specific refresh rate can be calculated, so input is 50/400, rather than `0 - 70` as used in Companion and mentioned in the MPM docs. 

https://github.com/EdgeTX/edgetx/blob/b74cd10927f20f919a2dcd697732e0b01cf45c2f/radio/src/gui/colorlcd/mpm_settings.cpp#L99-L106

So in that particular instance, `setGetValueHandler` doesn't appear to be working?

And I guess the generic handler here isn't working either...

https://github.com/EdgeTX/edgetx/blob/b74cd10927f20f919a2dcd697732e0b01cf45c2f/radio/src/gui/colorlcd/mpm_settings.cpp#L120

eshifri, 2022-08-13:
rfProto->getOptionStr() returns null.
https://github.com/EdgeTX/edgetx/blob/b74cd10927f20f919a2dcd697732e0b01cf45c2f/radio/src/gui/colorlcd/mpm_settings.cpp#L60

pfeerick, 2022-08-13:
Are you trying to run this in `simu`? This code executes in response to status of the MPM, and `simu` doesn't provide any emulation for that, so will always 'fail'. 

with the following changes

```diff
diff --git a/radio/src/gui/colorlcd/mpm_settings.cpp b/radio/src/gui/colorlcd/mpm_settings.cpp
index 7da50159c..d51655649 100644
--- a/radio/src/gui/colorlcd/mpm_settings.cpp
+++ b/radio/src/gui/colorlcd/mpm_settings.cpp
@@ -59,9 +59,10 @@ void MPMProtoOption::update(const MultiRfProtocols::RfProto* rfProto, ModuleData
 {
   if (!rfProto || !rfProto->getOptionStr()) {
     lv_obj_add_flag(lvobj, LV_OBJ_FLAG_HIDDEN);
+    TRACE("MPMProtoOption::update LV_OBJ_FLAG_HIDDEN");
     return;
   }
-
+  TRACE("MPMProtoOption::update %s", rfProto->getOptionStr());
   lv_obj_clear_flag(lvobj, LV_OBJ_FLAG_HIDDEN);
 
   const char *title = rfProto->getOptionStr();
@@ -100,6 +101,7 @@ void MPMProtoOption::update(const MultiRfProtocols::RfProto* rfProto, ModuleData
     edit->setMin(50);
     edit->setMax(400);
     edit->setGetValueHandler(GET_DEFAULT(50 + 5 * md->multi.optionValue));
+    TRACE("MPMProtoOption::update FS_AFHDS2A setGetValueHandler(%d)", md->multi.optionValue);
     edit->setSetValueHandler(SET_VALUE(md->multi.optionValue, (newValue - 50) / 5));
     edit->setStep(5);
     lv_obj_clear_flag(edit->getLvObj(), LV_OBJ_FLAG_HIDDEN);
@@ -118,6 +120,7 @@ void MPMProtoOption::update(const MultiRfProtocols::RfProto* rfProto, ModuleData
       edit->setMin(-128);
       edit->setMax(127);
       edit->setGetValueHandler(GET_DEFAULT(md->multi.optionValue));
+      TRACE("MPMProtoOption::update setGetValueHandler(%d)", md->multi.optionValue);
       edit->setSetValueHandler(SET_DEFAULT(md->multi.optionValue));
       lv_obj_clear_flag(edit->getLvObj(), LV_OBJ_FLAG_HIDDEN);
       lv_obj_clear_flag(rssi->getLvObj(), LV_OBJ_FLAG_HIDDEN);
```

I get 

```
79.01s: MPMProtoOption::update Servo update rate
79.01s: MPMProtoOption::update FS_AFHDS2A setGetValueHandler(0)
79.04s: MPMProtoOption::update Servo update rate
79.04s: MPMProtoOption::update FS_AFHDS2A setGetValueHandler(0)

```
when selecting the FlSky2A protocol, and field is blank. If I increment it two values

```
90.03s: !protos->isScanning()
90.27s: MPMProtoOption::update Servo update rate
90.27s: MPMProtoOption::update FS_AFHDS2A setGetValueHandler(2)
```

(note: only one call to setGetValueHandler this time)

if I go into DSM

```
384.83s: INDEV_STATE_RELEASED
384.97s: MPMProtoOption::update Enable max. throw
385.00s: MPMProtoOption::update Enable max. throw
```

if I go into Frsky
```
413.99s: INDEV_STATE_RELEASED
414.13s: MPMProtoOption::update RF Freq. fine tune
414.13s: MPMProtoOption::update setGetValueHandler(0)
414.16s: MPMProtoOption::update RF Freq. fine tune
414.16s: MPMProtoOption::update setGetValueHandler(0)
```

So `MPMProtoOption::update` is running, but the edit field isn't updating. Looks like need to set the segger up again...

eshifri, 2022-08-13:
Bummers... Completely forgot that MPM provides capabilities.
So this screen on simu is different from the real TX?

I think something is wrong with NumberEdit there: scroller does not change the value - only touch screen variant works (on TX, this time :-) )

eshifri, 2022-08-13:
"There's the rub!":
in the NumberEdit constructor:
  if (_getValue) {
    auto value = _getValue();
    this->setValue(value);
  }
so if the getValue handler is not set nothing is displayed.

If I hijack the code in the mpm_settings.cpp by moving the edit constructor into the update() (and add box to the class) everything works:
  {
    auto edit = new NumberEdit(
        box, rect_t{}, 50, 400, GET_DEFAULT(50 + 5 * md->multi.optionValue),
        SET_VALUE(md->multi.optionValue, (newValue - 50) / 5));
    edit->setStep(5);
    lv_obj_clear_flag(edit->getLvObj(), LV_OBJ_FLAG_HIDDEN);

I doubt this can be a solution, but it explains the behavior.

pfeerick, 2022-08-14:
Yeah, unforutnately simu / simulator is different to the radio in this respect since it only simulates the radio firmware, not external stuff.

Oh, nice. That actually makes sense... the get and set handler is being set, *but the current value isn't being set*! 🤦 i.e. 

```diff
diff --git a/radio/src/gui/colorlcd/mpm_settings.cpp b/radio/src/gui/colorlcd/mpm_settings.cpp
index 7da50159c..e75af2683 100644
--- a/radio/src/gui/colorlcd/mpm_settings.cpp
+++ b/radio/src/gui/colorlcd/mpm_settings.cpp
@@ -99,6 +99,7 @@ void MPMProtoOption::update(const MultiRfProtocols::RfProto* rfProto, ModuleData
   } else if (rfProto->proto == MODULE_SUBTYPE_MULTI_FS_AFHDS2A) {
     edit->setMin(50);
     edit->setMax(400);
+    edit->setValue(50 + 5 * md->multi.optionValue);
     edit->setGetValueHandler(GET_DEFAULT(50 + 5 * md->multi.optionValue));
     edit->setSetValueHandler(SET_VALUE(md->multi.optionValue, (newValue - 50) / 5));
     edit->setStep(5);
@@ -117,6 +118,7 @@ void MPMProtoOption::update(const MultiRfProtocols::RfProto* rfProto, ModuleData
     } else {
       edit->setMin(-128);
       edit->setMax(127);
+      edit->setValue(md->multi.optionValue);
       edit->setGetValueHandler(GET_DEFAULT(md->multi.optionValue));
       edit->setSetValueHandler(SET_DEFAULT(md->multi.optionValue));
       lv_obj_clear_flag(edit->getLvObj(), LV_OBJ_FLAG_HIDDEN);
```

IMO it's not the job of the handlers to auto-fire when initalised, so a few things like this are needed. I'm in the process of going through this now as there are a few more places this is needed, as well as some other UI cosmetics when stuff is being shown when not relevant, etc.  Thanks for finding it! :)

PR #2209 (2022-09-05): Fixes for MPM UI for colorlcd
Resolves #2203

Summary of changes:
- set field values to current option value when configuring fields
- hide sub-protocol choice if the list is empty
closed 2022-09-05 by pfeerick