EdgeTX/edgetx #2204: Broken custom curves.

ZZ-Cat, 2022-08-12:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

When a custom curve is edited, the curve immediately overrides any previous values in the curve with random values, & another random standard curve is automatically placed below the curve that has been edited.

[image]
[image]
**Above:** Broken curves as they appear in the Curves Menu. CV4 is randomly generated on editing the custom curve above.

[image]
[image]
**Above:** Broken curves as they appear in their respective editors. Note how CV3's values are overridden with random numbers.

### Expected Behavior

Any values that are set in a custom curve by the user is meant to be left unaltered, unless the user has made changes to the curve, themselves. No additional curves should be automatically added either.

[image]
**Above:** Correct custom curve as it is intended to appear in the curves menu.

[image]
**Above:** Correct custom curve as it is intended to appear in its respective editor.

### Steps To Reproduce

**Preparation:**
1. Flash your controller with any EdgeTX 2.8.0 Nightly release from d708285 (June 2, 2022) onward.

**Method 1 - In your controller:**
1. In the Model Settings, go into the Curves tab.
2. Pick an empty curve slot, any curve slot; & go into its editor.
3. Under Type, select Custom.
4. This bug rears its ugly head instantaneously.
5. When you exit out of the curve's editor, you will also see an additional standard curve that has been automatically placed below the one you just created/edited.

**Method 2 - From EdgeTX Companion:**
1. Create a custom curve as you normally would in Companion.
2. Save your model settings to your controller's uSD card & put that in your controller.
3. Power up your controller.
4. In the Model Settings, go into the Curves tab & find the custom curve that you have created.
5. All looks good & fine, right? Perfect. Now, edit that curve.
6. _Now,_ you see the bug & all its nasty glory.
7. Just like in Method 1, when you exit out of the curve's editor, you will see an additional curve that has been placed below the one you just edited.

### Version

Nightly (Please give date/commit below)

### Transmitter

Radiomaster TX16S

### Anything else?

The last known nightly release where this bug wasn't present is 60fbe5e from May 5, 2022.
This bug is present in every nightly release from d708285 of June 2, 2022 all the way up to the latest nightly release (basically, since the rewrite of the user interface).

This does not affect Standard curves. It only affects Custom curves.

**Before anyone asks:** The uSD card I have in my RadioMaster is a SanDisk Ultra 8 GB class 1 micro SDHC card. It's not the same as what my RadioMaster came with. I yeeted the old uSD card, when I converted my RadioMaster to EdgeTX (I've been here since 2.4.0 days). I am aware that some folks have had issues with the uSD card that their controller came with. I dodged _that_ bullet the very second I heard about this.

**Additional information:** I don't have any mods in my RadioMaster, aside of the fact that it's the Mark 1 Max Edition & I have upgraded it to the AG01 gimbals. I highly doubt this has _any_ effect on what I'm seeing here. But, hey... stranger things have happened.

pfeerick, 2022-08-12:
I can alleviate your concern about SD card and radio hardware immediately, as this is reproducible on a recent build of the Companion radio simulator ;) So this will solely be around the LVGL re-write and some gremlin it introduced (it brought plenty of friends with it).

ZZ-Cat, 2022-08-12:
> I can alleviate your concern about SD card and radio hardware immediately, as this is reproducible on a recent build of the Companion radio simulator ;) So this will solely be around the LVGL re-write and some gremlin it introduced (it brought plenty of friends with it). 

Thanks for that. =^/.^=

PR #2229 (2022-08-24): fix: custom curves corruption
Fixes #2204
commit 0e8f4e55 (2022-08-24): fix: custom curves corruption

Fixes #2204
closed 2022-08-24 by pfeerick