EdgeTX/edgetx #2212: Retain Page Options when Changing Layout

jmxp69, 2022-08-15:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

This may be completely preempted with LVGL, but I'm going to ask anyway.

When changing the layout for a main view, the options Top Bar/Flight Mode/Sliders/Trims are reset. Can we please not change those just because the layout is adjusted? Very annoying while working with widget layouts.

### Describe the solution you'd like

Do not change Top bar, Flight mode, Sliders, Trims state when layout is changed.

### Describe alternatives you've considered

Ethos? :)

pfeerick, 2022-08-15:
Mere mention of Ethos could get this issue closed without resolution :-P

LVGL doesn't change anything here, as evidenced in the nighties. As the options are actually part of the layout. i.e. this is why the 2x4 layout has some extra options not present on any other layout ... this may require some redesign of the layout system. I do agree it would be nice if they stayed as set if common to other layouts.

jmxp69, 2022-08-16:
While we're on this subject, I noticed on 2.7.1 if I switch to a telemetry page, long press the tele button, the page editor brings me to the page I had been viewing. On 2.8.x that behavior changes and the editor always goes to page 1. Long pressing the tele button to launch the page editor on the page being viewed is the correct behavior IMHO.

jmxp69, 2022-08-16:
> Mere mention of Ethos could get this issue closed without resolution :-P
Just a little shock value to wake up the audience.

pfeerick, 2022-08-16:
> While we're on this subject, I noticed on 2.7.1 if I switch to a telemetry page, long press the tele button, the page editor brings me to the page I had been viewing. On 2.8.x that behavior changes [...]

Yeah, that's a regression... earlier versions used to behave like 2.8 does now... and I think it was 2.7 that made it jump to the tab for the current view. Thanks for catching that.

> Just a little shock value to wake up the audience.

More like showed signs of insanity even contemplating it... Ethos is pure evil since it's Frsky ... you know better! 😝

PR #2218 (2022-08-24): View main fixes
Fixes #2212 (partly)

Summary of changes:
- Screen setup now starts with the tab corresponding to the active view.
- Active view is saved and restored.
PR #5530 (2024-09-14): feat(color): keep screen options when changing layout.
Fixes #2212 

Save and restore the screen options for topbar, flight modes, sliders, trims and mirrored when the layout is changed.

Note: does not apply if either the old or new layout is App Mode.
closed 2024-09-14 by pfeerick