EdgeTX/edgetx #2213: [TX12] Remote control channel settings issue

lida2003, 2022-08-15:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

1. Under the [mixs] page, assign SB to ch6 through the roller; edgetx 2.7.1 doesn't work and opentx 2.3.11 works.
2. Under the [logical switch] page, use the SA key to assign V1 and V2; edgetx 2.7.1 doesn't work and opentx 2.3.11 works.

Anything or way of channel settings changed in edgetx?

### Expected Behavior

edgetx does the work as opentx 2.3.11 did.

### Steps To Reproduce

1. Flash TX12 with default configuration (OpenTx 2.3.11 or EdgeTx 2.7.1)  (or remove all channel setting except pitch/yaw/roll/throttle)
2. go to  [logical switch] page, add L01 --> sticky, SA(down) to V1, V2  (OpenTx select V1 and V2, press SA button to do the work)
3. go to  [mixs]  page, assign SB to ch6 (OpenTx use roller to select and save the settings; EdgeTx doesn't find SB)

Note: Step 2/3 EdgeTx doesn't work as OpenTx did.

### Version

2.7.1

### Transmitter

RadioMaster TX12 / TX12MK2

gagarinlg, 2023-09-18:
is this still an issue with 2.9?

lida2003, 2023-09-18:
> is this still an issue with 2.9?

I didn't see any other feedback about this issue. I'm using OpenTx 2.3.11.

closed 2025-02-22 by github-actions[bot]