EdgeTX/edgetx #2214: Model name issue V2.8 nightly

chucky-whiskers, 2022-08-16:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Creating new models and modifying existing models does not allow for name change to take effect on "models" page. When changing the model name in "model settings", the name does not change in the "model selection" page. Duplicating this model will create a duplicate with the appropriate name, but upon deleting the original unnamed model or powering off the transmitter, the model reverts to the base Model 'X' name. Widget settings are also acting the same way.

### Expected Behavior

Creating a new model should allow the model to be named upon creation, in my opinion, but it should be able to be renamed permanently in Model Settings page. Widget settings changes should not revert to the settings from the time of boot up or model selection upon choosing a different model or powering off radio. 

### Steps To Reproduce

1. Create new model.
2. Open Model Settings via MDL button
3. Rename model
4. Set up widget
5. Power off transmitter
6. Power on transmitter
7. Edit Widget Settings
8. Open Model Selection
9. Duplicate model
10. Select properly named model
11. Delete un-renamed model
12. Power off transmitter
13. Power on transmitter
15. Check Model Selection screen to see the original Model 'X' name
16. Check Widget Settings to see original, unedited settings

### Version

Nightly (Please give date/commit below)

### Transmitter

Radiomaster TX16S

### Anything else?

Date of issue is Aug. 15, 2022 for reference on Nightly version.

eshifri, 2022-08-16:
"Duplicate" writes a model name as "" into the models.yml.
This is due to the constructor:
https://github.com/EdgeTX/edgetx/blob/7800a6de422a442485531ee00dcbe00c8451fea7/radio/src/storage/modelslist.cpp#L39

I think modelName should be send at least down to 
https://github.com/EdgeTX/edgetx/blob/7800a6de422a442485531ee00dcbe00c8451fea7/radio/src/storage/modelslist.cpp#L211

eshifri, 2022-08-17:
Will it be OK to send ModelCell& rather than model file name into this code?

pfeerick, 2022-08-17:
I think we need to start at the start here - I don't think the name being entered in the model setup page is being registered, so of course it's not going to go anywhere after that if it doesn't even make it that far. Try changing the name of a model via model setup. Then go back into again... name is still there. Then cycle `simu`/transmiter... name is back to the default model name, since it seems that `ModelNameEdit` never triggered its change handler... 

Having had a quick look at the code in comparison to how it looked in 2.7... I would hazard a guess it's not using the LVGL callbacks to trigger the update code... but I am only guessing at this point...

eshifri, 2022-08-17:
I have a fix for "duplicate", but it is difficult to test it since "edit" does not work.
The problem wit edit the callback function
https://github.com/EdgeTX/edgetx/blob/6028c24f360ff99a6a8e95cc6187914f7af8762b/radio/src/gui/colorlcd/model_setup.cpp#L47
is never called. 
I'm not sure it is related to lvgl: the changeEnd() that is supposed to be called from onEvent() of the text edit is never called as well, since the events are re-purposed. 
Also just a speculation at this point.

eshifri, 2022-08-18:
And actually it fixed this problem as well. :-)

pfeerick, 2022-08-18:
Seriously? Nice. I'll double-check and close if that's so... as that is a nice curly set of steps to follow... thanks for the heads up! :)

pfeerick, 2022-08-20:
Ok, so most of this issue has been resolved... there just remains one peculiarity related to the duplication - where I think the `models.yml` model name is not using the model name when duplicating:

1. Duplicate an existing model. Both will show the same name as original model  in the model select screen, and in model setup screen.
2. Power off transmitter / Power on transmitter (or alternately, select another model)
3. Check Model Selection screen to see duplicated model name is now model## (where ## is an incremental number)
4. Check Model Setup to see the model still has the name set

eshifri, 2022-08-20:
Hm, I think this may be considered a feature - not a bug. :-)

pfeerick, 2022-08-20:
lol... yeah, I'd like to describe it as that... as I understand the behaviour... but when there's inconsistency, I call bug! :bug: Basically after the other, more insiduious bug was dealt with, we are left with what you pointed out here... https://github.com/EdgeTX/edgetx/issues/2214#issuecomment-1216174868 ... did you end up patching just that behaviour?

eshifri, 2022-08-20:
Yes. My fix was to add model name as a new parameter to the constructor and one or two other functions.
I can prepare a PR if you think it is worth fixing.

closed 2022-08-26 by pfeerick

chucky-whiskers, 2022-10-11:
I would prefer model name because I have about 30 different models and
trying to figure out which is which in each of 5 folders is confusing.

PR #2224 (2022-08-26): Write model name into models.yml when duplicating a model (#2214)

Fixes #2214

Summary of changes: add model name as a parameter to several functions to make it available to the ModelCell constructor.