EdgeTX/edgetx #2215: Use GV name if assigned instead of GVx

jmxp69, 2022-08-18:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

If we edit a global variable GVx and give it a name: FLT then add that global variable as a widget using Value that global variable is displayed on the screen as GVx and not FLT. 

### Describe the solution you'd like

I believe if a user assigns a name to a GV, the name should be used throughout. We do the same thing with Curves. If we use no curve name, CVx is used. If we assign a name, that name is used throughout.

### Describe alternatives you've considered

Buying pfeerick a stack raffle tickets for EdgeTX Fest giveaways.

pfeerick, 2022-08-19:
I'm afraid you'd probably have to do it in person, with a bunch of disguises and false names... given I'm part of the project my name won't be in the draw ;) 

Oh, interesting... the name *is* shown when it's a GV only list ...
[image]

... but not when it's a source list ... interesting, and not very consistent! 😆 
[image]

pfeerick, 2022-08-19:
This just seems to leave it so you need to see an un-named one before you realise they're the GVs...

[image]

gagarinlg, 2022-08-19:
> I wonder if it would be better to show like with Companion in this instance... i.e so you know it's a GV...

This is what I would opt for, but is there enough screen space on the B&W radios?

pfeerick, 2022-08-19:
Yeah, that's what I was thinking afterward, so am going to avoid that problem for now. 😆 On B&W they'll just have to continue to suffer with `GV#` as the UI there changes in place, rather than via a popup list.

jmxp69, 2022-08-19:
> I'm afraid you'd probably have to do it in person, with a bunch of disguises and false names... given I'm part of the project my name won't be in the draw ;)

AWWW - I'll just have to find my kryptonite elsewhere I guess. 

> 
> Oh, interesting... the name _is_ shown when it's a GV only list ... [image]
> 
> ... but not when it's a source list ... very interesting indeed, and not very consistent! 😆 [image]
> 
> Hm... this looks like where to fix it... and the nice thing is it should apply to B&W and Colorlcd radios if done there... I wonder if it would be better to show like with Companion in this instance... i.e so you know it's a GV...
> 
> [image]
> 
> https://github.com/EdgeTX/edgetx/blob/6648dffc8c89d34fd26aba544e507d3cf3b20e47/radio/src/strhelpers.cpp#L677-L678

Why does your curve 2 and 3 have 4 characters hmmmm????? I saw that....

gjhaynes, 2022-08-19:
Hi Peter, I don't mind if the user-defined name is prefixed with GV: when it appears in a drop down list, but would prefer to display only the user-defined name when it is used in a widget for example. At that point I don't need to know that the displayed value's source is a GV, I just want it to have a meaningful label. 
Cheers, Geoff.

PR #2219 (2023-11-12): feat(color): Show index and name (when set) for GVs in input source
Resolves #2215

Summary of changes:
 * on colorlcd, when showing the sources list, if GV has a name, show its index and its name, as opposed to just its index. 
 
 This is just proof of concept... the code is still quite ugly... 
 
Before: 
[image]

After:
[image]

I'm pushing this to 2.9 as I want to piggyback and extend this on top of #2085, and also show names of LS when present.
closed 2023-11-12 by pfeerick