EdgeTX/edgetx #2232: Add LI-ION battery voltage table in BATT ANALOG Widget

DG9BFC, 2022-08-23:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

offer shmuely added the option to switch from Li-Po to Li-Ion in the Batt check widget
but not in the BattAnalog widget

### Describe the solution you'd like

i would like to have that same option in BattAnalog cause i use Li-Ion cells in my Tx16s

### Additional context

The cell percentage table in the BattCheck widget needs some finetuning cause the table now is linear over the range 3 to 4.2v

offer-shmuely, 2022-08-23:
I am adding to both 2 widget, the option to choose Li-ion voltage range
we need to agree on the values

I suggested this one

[image]

DG9BFC, suggested:
blue is your curve (more or less linear drop of voltage)
orange is what i would use (your is 0% at 3v and mine still shows 9% at 3V)
i guess my curve fits a bit better ... what do you think?

[image]

offer-shmuely, 2022-08-23:
does anyone have more input?

DG9BFC, 2022-08-23:
with very low  discharge current (0.2c or even lower) the curve may 
[image]
be flatter (hovering at 3,6 or 3.7) to drop even faster at the discharge end

DG9BFC, 2022-08-23:
above curve is what i found in a youtube video (and made a screenshotfrom it)

DG9BFC, 2022-08-23:
https://www.batteryspace.com/prod-specs/NCR18650B.pdf

and here cell data of an 18650

and grabbed the discharge cuve from there you see;
[image]

DG9BFC, 2022-08-23:
if it is a bit too much work for you i can try to make that curve and send you that part of the widget (just drop me an email addy where to send it)

pfeerick, 2022-08-24:
The biggest problem here is going to be no curve is going to be 100% right for a given application or battery- as the behaviour is both battery and load dependent. i.e. the curve DG9BFC shows is [typical for Panasonic NCR18650B](https://lygte-info.dk/review/batteries2012/Panasonic%20NCR18650B%203400mAh%20(Green)%20UK.html)...
[image]

And also [Samsung INR18650-20R](https://lygte-info.dk/review/batteries2012/Samsung%20INR18650-20R%202000mAh%20%28Green%29%20UK.html)
[image]

But then you get the [Samsung INR18650-30Q](https://lygte-info.dk/review/batteries2012/Samsung%20INR18650-30Q%203000mAh%20%28Pink%29%20UK.html)  and [Sony US18650VTC6](https://lygte-info.dk/review/batteries2012/Sony%20US18650VTC6%203000mAh%20%28Green%29%20UK.html), which have less of a drop off curve, as they reach end of capacity...
[image]
[image]

And as you can see from some of the charts, 3.0v is right in the middle of the battery avalanche point, especially at lower loadings as might be the case if it's in use in a transmitter! So I think at best all that can be provided is a rough curve, that reflects that a battery voltage will initially drop, and then have a steady decline and a rapid dropoff. It's by no means a linear line, as has been crudely used in the past by even some manufacturers. 

Also... I'm going to move this issue, as it belongs in the sdcard repo, since it is specifically about a widget included in the sdcard pack.