EdgeTX/edgetx #2236: Enable USB-CDC (CLI, Telemetry passthrough) by default

r1000ru, 2022-08-26:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

All multiprotocol radios can be modified into ELRS. It would be great if USB-CDC was enabled by default for all hardware.
Even if there are no plans to change the internal module, USB-CDC can be useful for telemetry mirror

### Describe the solution you'd like

Add options in build system

raphaelcoeffic, 2022-08-28:
As far as I can see, USB-CDC is enabled for every target. `CLI` isn’t though. That doesn’t prevent usage of “Telemetry Mirror” over USB though.

pfeerick, 2022-08-29:
It would be nice if specific cases could be mentioned - i.e. if requesting it enabled by default, can you say what instance of it not being enabled made you ask for it to be enabled by default? 😆