EdgeTX/edgetx #2239: Print logviewer graph

MicheleVilla, 2022-08-28:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Hi
In Companion logviewer  would be very useful a function that allows you to print the displayed graph
actually need to capture screen or export csv file in libreoffice calc

### Describe the solution you'd like

add a print button in logviewer form

MicheleVilla, 2022-08-29:
Alternatively you could export as an image

elecpower, 2023-07-24:
> Alternatively you could export as an image

You could just use a screen capture utility if all you need is an image

closed 2026-04-20 by elecpower