EdgeTX/edgetx #2246: FlySky NV14 layout files

nightflasher, 2022-08-29:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

the screen layouts are mostly for widescreen radios, but the nv14 has a portrait screen, so the screen layout selectors are a bit misleading and can be some kind of disappointing. 

### Describe the solution you'd like

include portrait mode icons (made them already in 20x25px)
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

raphaelcoeffic, 2022-08-30:
I believe these should be computed rather than included as fixed bitmaps. Should would allow for more flexibility. In general, it seems to me that the screen layouts for NV14 are not very well adapted in general. This is something we should revisit probably.

richardclli, 2023-03-22:
Seems this is related to https://github.com/EdgeTX/edgetx/pull/3282
Is this PR solves the problem properly? @nightflasher

nightflasher, 2023-03-22:
[image]
No, it doesn't work. It's only a ratio thing.

philmoz, 2023-03-22:
> No, it doesn't work. It's only a aspect ratio thing.

Are you sure you tested the right build?
It should look like this on the NV14:
[image]

nightflasher, 2023-03-22:
EdgeTX 2.8.1 - which one should it be? Also, my NV14 still identifies itself as EL18

pfeerick, 2023-03-22:
main branch, so a nightly. Or wait for 2.9 ;)

pfeerick, 2023-03-22:
Resolved by https://github.com/EdgeTX/edgetx/pull/3282

closed 2023-03-22 by pfeerick

nightflasher, 2023-03-22:
oh okay - there is no nightly for the NIRVANA :'(

pfeerick, 2023-03-23:
On Buddy, you'll be after the Flysky EL18 - make sure you've ticked off show pre-releases under filters if you want to see nightlies. 

[image]

The firmware is identical for the EL18 and NV14, it auto-detects which hardware it is on at runtime, and makes some adjustments accordingly. Only gotcha is the firmware will in some places state it is the el18 (i.e. because that is the firmware file that was flashed, not the radio model that is being displayed).