EdgeTX/edgetx #2248: Sound bug

Billy60, 2022-08-29:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

The Alt value is not spoken on th eloudspeaker : a mumble is the only sound instead of the altitudes in meters.

### Expected Behavior

Should speak out altitudes in m according to the variometer and Altmeter.

### Steps To Reproduce

For any Alt values, I get only the same mumble sound.

### Version

2.7.1

### Transmitter

Radiomaster TX16S

pfeerick, 2022-08-30:
Do other audio prompts sound ok? (i.e. "Welcome to EdgeTX" when you turn it on?) Thus hopefully ruling out issues with volume settings and speaker?

Billy60, 2022-08-30:
The mumble is only for the Alt values. Other sounds are ok.