EdgeTX/edgetx #2250: Telemetry reset error Horus x12s and FrSky Neuron ESC

oysteinbach, 2022-08-30:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

I am struggling with the following.
Frsky Horus x12s running EdgeTX v 2.7.1
Frsky Neuron 80 connected to X8R rx.
Software on rx and ESC is updated to the latest version.
There are 2 problems here.
Even if the motor doesn't run the ESC report 19.3A in the telemetry.
When both tx and rx is switched off the telemetry doesn't reset. And even if I use a Global function on the "trainer switch" the telemetry values are reset to 0 when I hold the button. Once the button switch back the values are "restored" for mAh usage and the current report approx. 19.3A even if the motor is not running.

### Expected Behavior

When switching telemetry reset the values should start from zero.
When motor is not running, the ESC for sure doesn't draw 19.3A

### Steps To Reproduce

Mentioned in current behavior

### Version

2.7.1

### Transmitter

FrSky X12

raphaelcoeffic, 2022-08-30:
@oysteinbach which sensors are listed when discovering sensors?

You should be able to see the voltage as well. Is it consistent?

oysteinbach, 2022-08-30:
Hi!
The voltage seems to be correct. The total consumption in mAh isn't reset either. Right now I am struggling with the Neuron 80 and Horus x12s as tx. I do also have possibility to check if it works better with Radiomaster x16S if that helps.
Let me know if you need any other info.

oysteinbach, 2022-08-30:
Frankly speaking, I haven't double checked the voltage with a multimeter to compare, but it looks OK...

raphaelcoeffic, 2022-08-30:
> I do also have possibility to check if it works better with Radiomaster x16S if that helps.

It won't be any different: it's the same code.

raphaelcoeffic, 2022-08-30:
Ideally, we would need a dump of the data on the `S.PORT` line output by this ESC (with some details on voltage/current to be able to counter-check). For my part, I don't have that ESC (nor any of the series).

oysteinbach, 2022-08-30:
Is there something I can do to assist?

oysteinbach, 2022-08-30:
I have posted this issue on the FB group as well. And others have experienced the same. Swapped back to OTX and it is working fine. I think that hurts.... :-)

oysteinbach, 2022-08-30:
Is there any way to get a dump of the s.port data?

raphaelcoeffic, 2022-08-30:
> Is there any way to get a dump of the s.port data? Or do you need a special hardware dongle to do that?

Seems like I found someone able to do just this ;-)

oysteinbach, 2022-08-30:
I have tested to have the telemetry reset function both as a global variable and also individual in the model setting. Didn't do any difference.

3djc, 2022-08-30:
I've tried to duplicate the issue, so far without success. Tx16s + X8R + Neuron 60

[image]

It uses a multi  but that should not make a difference, I'll try with a x7 to use an xjt, reporting here soon

3djc, 2022-08-30:
Same goes with X7

[image]

oysteinbach, 2022-08-30:
I'll connect the Horus and the x8r to another Neuron 80 and retest. I do also have a Neuron 60 new in the box. Need to solder some connectors and connect a brushless first. 
TBC.

oysteinbach, 2022-08-30:
So I have connected another Neuron 80A ESC that I have to a S8R receiver and used my Tx16S radio. I don't have the Horus available this afternoon. And the tx16s works as intended.
I doubt that it is the receiver, so we can look to either ESC or TX. The ESC is kind of stupid in this sense, so personally I believe it must be the difference of the Horus x12s and the Radiomaster tx16s. In my post in FB "EdgeTX Users" one replies about the same problems with Horus and telemetry, so he switched back again to OTX to make it work. If I have to, I must do it, but it hurts to leave Edgetx back to OTX.

gagarinlg, 2023-09-18:
Is this still an issue?

3djc, 2023-09-18:
tx16s and Horus share the same telemetry code, it would be very strange that one sees it, the other not

oysteinbach, 2023-09-18:
Ok, I'll check it out.
At least it didn't work last time I tried.
Any extra logging possible? A hidden code to activate it?