EdgeTX/edgetx #2260: The radio shows some models but companion reads and sees other

luzrael, 2022-09-02:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Other (Please specify below)

### Current Behavior

I had OpenTX installed, I updated to EdgeTX 2.7.1
All the models converted successfully. 
Next I read the models and settings from the EdgeTX companion. 
In the companion I modified the models and changed the name of one of them (from LanYu to LanYuNew). 
To finish, I wrote the models and settings to the radio.
I turn the transmitter off and on.  At the transmiter I still see the old models.
If I reread the models and the settings from the companion, the models are well updated and the one with the changed name is also there, but in the transmiter it shows me the old models.

### Expected Behavior

See a have the same models in the radio and in the companion.  The models that companion reads and saves on the radio are not the ones on the radio

### Steps To Reproduce

1. I Read models and settings from radio to the companion.
2. I modify a model and rename another in the companion.
3. Write models and settings from companion to the radio.
4. On the radio I dont see the renamed model nor the changes. 
5. I Read models and settings from radio again to the companion.
6. In the comanion, the read models are updated and one is renamed, but in the radio it only shows the old ones.

### Version

2.7.1

### Transmitter

Radiomaster TX16S

### Anything else?

[image]
[image]

pfeerick, 2022-09-02:
This most likely means you have model files on the radio that are not actually being used - i.e. perhaps still have the `.bin` version of some of those models on your radio? Anyway, this bug will most likely be gone in a current nightly build of either the firmware, Companion, and will certainly be blown away when the new label-based model select screen gets merged in the next few days.

luzrael, 2022-09-02:
Thanks for the answer.

There is no old bin file.  

The problem is simpler.  

I delete ALL the models (*.yml) from the radio card.
I turn on the radio. In the radio there are NO models or templates available.  
I connect the USB cable to the radio, and start the companion.
From the companion I read the models (documentX.etx).
In the companion (documentsX.etx) I see that the radio has the new models that I had created, modified and renamed.
But the radio shows that it does not have any.

If I write these models to the radio (documentsX.etx), the still shows that it has no models.

Please, Ican do something to use the transmitter? I do not have models visible in the radio, or if I copy the yaml files to the SD card, I see/have the old models that do not work for me.

Thanks

eshifri, 2022-09-02:
I think there is a chance it is fixed in the current nightly, but meanwhile you can manually edit the Models.yml file in the MODELS folder on the D card. It is a text file. When you connect you TX to the computer (no Companion) the SD card is visible as an external USB disk. You can see your models in the /MODELS folder. You can add/change visible models by editing the Models.yml file. (Obviously some backup will not hurt. :-) )

pfeerick, 2022-09-26:
I'm going to call this closed by 2.8 - as Companion should cleanup any "extra" models, and the new label-based system will also auto-add any models that aren't currently shown, so between the two of these this should be able to happen on colorlcd radios.

closed 2022-09-26 by pfeerick

luzrael, 2022-09-26:
Good evening,

I have found the version 2.8.0-RC1 software, as well as the contents of the
SD card.

But I can't find where the companion for version 2.8.0

It is available?

Can the companion of version 2.7.1 be used?

Thank you very much!

gagarinlg, 2022-09-26:
> Good evening, I have found the version 2.8.0-RC1 software, as well as the contents of the SD card. But I can't find where the companion for version 2.8.0 It is available? Can the companion of [...]

Please do not resurrect old issues. Here is the link, at the bottom of the page. 
https://github.com/EdgeTX/edgetx/releases/tag/v2.8.0-rc1

luzrael, 2022-09-26:
Sorry, I did not look well in the list of assets 🤦

luzrael, 2022-09-27:
Amazing!!!
Good job!!!

Both the new look and the change in organization of models is impressive.

Very well organized and very clean.

And the changes in the sizes of the letters and controls (especially the
dropdowns), make the screen a delight to use.

I have been impressed by the work done by the team working on EdgeTx.

Thank you very much for your time.

BTW, I am a programmer of drivers and real-time applications in C/C++ (and
hardly any assembler anymore) for Windows/Linux.

All this for applications in VoIP, analog and digital radios
(Tetra/Tetrapol) and management of PABXs in real time.

I've also made SQL/VBasic/REXX/COBOL compilers but it doesn't matter here
because LUA wouldn't know where to start.

I haven't looked at the EdgeTx source code, so I wouldn't know where to
start, but I would like to become a collaborator of the team that develops
EdgeTx.

I am going to see the indications that you have, and I would like to know
that if I do not understand something, I could ask you (or another
colleague could indicate me).

Thank you very much, I am very grateful 🙏

gagarinlg, 2022-09-27:
Just join the discord server and start coding!

Cattlebaron, 2024-11-07:
I have the same issue with my RadioMaster TX16S MK2. My desktop computer is a Mac running Mac OS sonoma 14.6.1. I have downloaded EdgeTX Companion 2.10.5 and successfully connect the the TX. Creating new models is not a problem as I have been using OpenTX for a decade. Once a model is created I save it to the TX and to the models folder on my computer. On switching the TX as at a flying field NO models show up to be selected. I have searched hi and lo for an answer without success, any guidance as where to look next?

Hank Deucker

luzrael, 2024-11-07:
> problem as qI have been using OpenTX for a decade. Once a model is created

Good morning,

The problem I had was that the models were not saved on the transmitter I
had connected.

The reason for the failure in my case is that Companion assumes that there
is a transmitter if it finds certain folders on an external drive.

So, if you have an external memory with the opentx folders, or an external
disk with those folders, and a USB drive with those folders, the Companion
program will save the models there and not on the connected transmitter.

That was my problem.

I solved it by removing the drives I used as a backup with the opentx
folders, before opening the Companion.

I don't know if that is your problem.

Greetings.

luzrael, 2024-11-08:
Hi,

One more question, to clarify the behavior of the Companion.

When the TX "connects" to the Companion, the Companion does not actually
connect to the TX.

And in fact the Companion does not even know if the TX is connected or not.

The only thing the Companion does is check that there is an external drive
that has certain folder names.

If that folder exists, the for the Companion "that folder" is the TX.

If you have more than one external disk, such as a USB drive (or two), with
this folders, the Companion assumes that those disks and folders are from
the TX.

When you press the save button, or reload models from the Companion, it
does NOT save or load the models to the TX.

What the Companion does is simply save, or load, the model files to a
directory that has the name models, or radio, or a couple more folder
names, that exists in the root of any of the computer's disks.

What Companion does is record and read files from the first hard drive that
has those folder names.

So if you have an external hard drive (a USB or any memory that shows up as
a hard drive) with some folders in the root of the drive that match radio,
models, etc. that is what Companion shows as the TX.

I hope this has been helpful, or has been able to clarify why Companion
shows that it has recorded the stuff on the TX and when you disconnect it
there is nothing on the TX.

Regards.

Cattlebaron, 2024-11-08:
Thank you for your prompt reply. I need to read the reply a number of times
to understand it as It discusses issues I'm no longer familiar with. So
some research is needed but I will reply in a couple of days.

Regards
Hank Deucker (AKA Cattlebaron)

pfeerick, 2024-11-08:
Basically Companion will look at each of drives connected to your computer, looking for one with the correct file system type and containing the 'RADIO' folder. As soon as it sees that, it thinks that is the SD card for your handset. So can be confused if you have more than one drive present with a RADIO folder. 

So check any other mounted drives, and make sure you don't have a RADIO folder at the top level... otherwise Companion maybe be reading/writing to there, not your handset. You should be able to double check this by simply reading without the handset plugged in... if it finds something, there's your problem! And if it doesn't, try reading, writing some change back, and reading back again... do the changes show up in Companion still? And then on the radio if you unplug it?

Cattlebaron, 2024-11-09:
Hi Peter,

Thanks for taking the time to send your email. After reading it a number of
times and following the suggestions I am still stuck.

here is what I have found:

   1. The only RADIO folder on my computer is RADIOMASTER/SDCARD/RADIO.
   2. "You should be able to double check this by..." results in 'Unable to
   find radio SD card'.
   3. Connecting the radio to the computer via the USB port results in the
   opening screen as expected and no matter how many times I press the MDL
   button or how long I hold it down no models are displayed for selection.
   4. Launching EdgeTX and selecting Read models from SD card will show a
   list of models after some time with the 'beach ball of death' routine. I
   can then do as I need to any model I select. Next I write models and
   settings to TX. Close EdgeTX and save to Model ETX files successfully. Then
   follows the 'beach ball of death' routine when Eject the folders as the
   final step.
   5. After all is closed down, then restart the radio as at the flying
   field we go back to step 3 above.

I feel that I am missing something very simple and obvious to most new
users but what that is I have no idea. It may be the prompts used that I am
not familiar with for instance when ordering food I use helpings but the
modern term is servings.

I don't know if my comments make sense, I sincerely hope so.

Cheers Hank

pfeerick, 2024-11-10:
So, to recap, am I right in understanding the following?

1) MacOS
2) RM TX16S MK2
3) EdgeTX 2.10.5 (handset and Companion)
4) Companion is not "detecting" a handset when it is not plugged in (so should not be incorrectly reading/writing elsewhere)
5) You can read and write from the handset, and any changes written are read back in Companion
6) Changes (and models created) are not reflected on the handset

Have you double checked you don't have a label selected which has no models assigned to it yet? i.e. I have 'nonehere` selected in this screenshot, with no models assigned to that label, so there is nothing to show. 

[image]

Can you upload a zip file of your RADIO and MODELS folder to github so we can see what is going on there. Look for the paperclip icon above the input/text edit box... don't attach via email reply as attachments are not accepted by github.

Cattlebaron, 2024-11-11:
I can only partially reply to your 6 points. I need help from my sons to
follow up on zip files.
 Now to your points

   1. 1 and 2 Definitely correct
   2. 3 Only partially, I downloaded EdgeTX 2.10.5 but have no idea what
   version is on the handset (I assume handset =TX) The TX was purchased late
   October and arrived early November.
   3. 4 I can connect the TX to my computer via USB. I.e. two folders show
   up on the Mac, after launching EdgeTX all appears good but I have no
   definitive way of knowing if the program is working or not.
   4. 5 I have not been able to do anything meaningful on the TX, this may
   be frustrating but I find it worse.
   5. 6 Any work I have done with EdgeTX is not displayed on the TX

Thank you for the .jpeg I have not had a screen as in the shown.

Regards Hank

pfeerick, 2024-11-11:
To determine the version, you can either go into Radio Settings/SYS -> Version, and see the version number listed. Or use the bootloader (hold the two horizontal trims inwards and power the radio on) and check the version number shown at the bottom of the screen (top version number is for the bootloader). 

Having mismached versions of the firmware to on the handset/TX to the version of Companion _can_ be problematic,  but if it is a recent purchase (and not old stock), that is less likely to be the problem here. 

For this point
> You can read and write from the handset, and any changes written are read back in Companion

I was more wanting to check if changes you make (and write) with Companion can be read back with Companion... suggesting that the problem is with the handset (e.g. firmware incompatibility / SD card issue / something else).

Cattlebaron, 2024-11-12:
I followed your advice and the version list is as follows:

FW: edgetx-tx16s
VERS: 2.7.1 (a55af02)
DATE: 2022-07-30
TIME: 14:34:00
CFGV: 221
OPTS: crossfire, ghost,afhds3, internalmulti, inte....
multimodule, luac

Unfortunately the photo missed the remainder of the word inte

I assume that version firmware can be updated, I will search the net and
hopefully find the information.

pfeerick, 2024-11-12:
This was the most important line:

VERS: 2.7.1 (a55af02)

Which most likely explains why you are not seeing the changes to models, as
2.10.x reads and writes model and radio settings quite differently.

If you download and run the 2.7.2 Companion, that *should *work for you.
Here is the direct link to the download, since I know you are on MacOS :
https://github.com/EdgeTX/edgetx/releases/download/v2.7.2/edgetx-cpn-osx-v2.7.2.zip

(note: the version difference is not accidental - v2.7.2 was solely an
update for MacOS Companion, due to some changes in MacOS 13 - and is
otherwise identical to v2.7.1).

If that at least gets things working for you again, you can then have a
look at
https://manual.edgetx.org/installing-and-updating-edgetx/update-from-opentx-to-edgetx-1
and
https://manual.edgetx.org/installing-and-updating-edgetx/update-from-an-earlier-version-of-edgetx-using-the-bootloader
for some guidance on how to do the update

Cattlebaron, 2024-11-12:
Again thanks for the advice. I think the best path for me is to upgrade the
firmware on the TX. I am familiar with the STM32CubeProgrammer as I use it
for H743 wing flight controllers. Further I will get assistance from a
computer expert. Can I get some guidance as to the firmware version I need
for TX please?

pfeerick, 2024-11-12:
Basically you just want the same version of the firmware as the version of
Companion you are using. So since you were wanting to use Companion 2.10.5,
you want firmware version 2.10.5 also.

Cattlebaron, 2024-11-13:
It looks like I have bricked my TX. I followed the instructions published
on
https://manual.edgetx.org/installing-and-updating-edgetx/update-from-an-earlier-version-of-edgetx-using-the-bootloader
but now nothing works. I chose the Flash option on the TX screen and it
reported back Successful. I opened the TX all seemed well but when checking
"Radio Settings/SYS -> Version" It reported back with VRES: 2.7.1
I switched off the TX and now it will not switch on again (the 6
switches light up and just keep the dance going). I know it is puzzling,
sorry to be a nuisance. I am sure that there will be a video about bricking
the TX.

Cattlebaron, 2024-11-13:
Very quick question, when flashing the firmware is all that is necessary to
use a text editor to change the version card number as the remaining
contents of the SD card are not important as no models have been created as
yet?

pfeerick, 2024-11-13:
No need to edit that file, it is not used at all, and is only there for
reference. You can even use a blank SD card and the radio will still boot
and work properly (although it will then demand that you calibrate the
sticks and pots, etc).

You may either have an issue with the RTC backup memory (necessitating
removal of the back cover, so you can pull the coin cell battery out and
let it reset), or have a MCU fuse setting that needs changing, per
https://manual.edgetx.org/installing-and-updating-edgetx/having-trouble-updating#i-updated-edgetx-and-now-my-radio-doesnt-turn-on-or-acts-strangely

Cattlebaron, 2024-11-19:
I am completely lost, I have made no progress. I have lost count of how many time I have flashed the firmware but I am greeted with a blank screen. I have done the RTC backup as described above. I will have to research MUC fuse setting and see how that goes. RadioMaster inform that the TX can address up to 32Gb micro SDcards. I will purchase one as 512Mb cards are no longer readily available if at all. I am confuse with the c480X272 File(?) when opened it has all folders for the SDcard. Is there a seperate file for defining the screen size?

pfeerick, 2024-11-29:
> I am greeted with a blank screen.

Probably need to check out the [troubleshooting section in the EdgeTX](https://manual.edgetx.org/installing-and-updating-edgetx/having-trouble-updating#i-updated-edgetx-and-now-my-radio-doesnt-turn-on-or-acts-strangely) manual if you haven't already - you probably need to change the fuse option bit if pulling the RTC battery didn't work and the radio is not booting up. 

>  I am confuse with the c480X272 File(?) when opened it has all folders for the SDcard. Is there a seperate file for defining the screen size?

The files are organised on the github repo by screen size, and then packed into a zip file. So by downloading/opening the c480X272 zip file, you have the files are are optimised for that screen size. If you were to grab one of the other zip files, the widgets and lua tools may be the laid out wrong for that screen size/orientation.

Cattlebaron, 2024-11-29:
Thanks for the tip! I have reached a level of frustration which has caused
me to get a warranty fix. I hope RadioMaster will agree to do this. I will
reread the recommended EdgeTX manual for the umteens time.

Cattlebaron, 2024-12-07:
Hi EdgeT masters,

I finally found time to act on your advice posted on 29/11/24. The good
news is that the radio now boots and I can access it with the latest
version of EdgeTX 2.10.5. I don't know how to help as I can no longer code
but I may be able to offer some help translating 'ginglish' into
aeromodeller language. Now on with getting it to work, would a blog help?