EdgeTX/edgetx #2261: Corrupted custom splash screen

ZZ-Cat, 2022-09-02:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

On startup, the splash screen is corrupted & appears to be dominated by "red-scale" (like grayscale, but instead of black-&-white, it's red-&-white).

I have included photos of my custom splash screen, plus four variants that I have created specifically to test this issue.
Fair warning: I couldn't get proper screencaps of the splash screen itself. So, "old fashioned screenshots" it is. 

[image]
**Above:** This is how my custom splash screen currently appears.

**Below:** I created these variations of my splash screen specifically to test this issue & they all appear similar to my splash screen above.
[image]
[image]
[image]
[image]

### Expected Behavior

Literally this...
[image]
**Above:** This is how my splash screen *should* be displayed.

**Below:** The same four variants of my splash screen, but this is how thar supposed to appear.
[image]
[image]
[image]
[image]

### Steps To Reproduce

#### Preparation:
If you're already using a custom splash screen, start from step 4 & go straight to testing.
1. Grab one of the custom splash screens that I have provided under "Resources for Testing".
2. Name your custom splash screen "splash".
3. Copy your splash screen over to the ```IMAGES``` folder in your uSD Card's directory.
4. Flash your controller with any nightly build from 5d399b0 (June 13, 2022) onward.

#### Testing:
1. Power up your controller.
2. ?
3. Blazing red splash screen!

### Version

Nightly (Please give date/commit below)

### Transmitter

Radiomaster TX16S

### Anything else?

#### Resources for Testing:
Here are a couple of example custom splash screens for testing this issue. These aren't necessary for you, if you're already using a custom splash screen.
[image]
[image]

#### Extras:
- The default splash screen is unaffected by this bug.
- This is another bug that has persisted since the user interface rewrite.
- This bug is present in every nighly build, since 5d399b0 from June 13, 2022 up to the latest nightly build.
- The last known nightly where this bug is not present is 33845b9 from June 6, 2022.

#### The bit on the side (miscellaneous trivia):
- The "blued-out" controller is a heavily color-graded photograph of my RadioMaster TX16S Max Edition (before I upgraded the gimbals).
- The custom splash screen that I'm currently using features an old concept logo that was suggested around about the time that the current EdgeTX logo was being designed. I liked the look of it, & it stuck ever since. =^/.~=

raphaelcoeffic, 2022-09-02:
Looks like this is caused by the LVGL image loader… at least the date you mention is suspicious in that regard.

pfeerick, 2022-09-02:
This sounds like the same issue that was being discussed [off topic in another PR](https://github.com/EdgeTX/edgetx/pull/2100#issuecomment-1209961108) - in that case if you re-saved the logo, but as 24 bit color (rather than 32bit) this issue went away. And it's not like you'd have any chance of seeing that color difference anyway with these screens...

raphaelcoeffic, 2022-09-02:
The broader question is probably: can we come up with some code mods that would effect proper color mapping?

ZZ-Cat, 2022-09-02:
> This sounds like the same issue that was being discussed [off topic in another PR](https://github.com/EdgeTX/edgetx/pull/2100#issuecomment-1209961108) - in that case if you re-saved the logo, but as [...]

=O/,..,O=
I just tried a 24-bit version of my splash screen & that worked. It shows up properly (not even the color banding, that I previously had, either).
Heh. The more ya know.

It seems weird that 32-bit depth images for the splash screen become corrupted like that, when they had previously worked with the old layout.

But, hey. Physically, I don't see any differences (that are significant enough to care about) between 32-bit or 24-bit on my RadioMaster's display panel, anyways. So, I'm happy to roll with 24-bit.

raphaelcoeffic, 2022-09-02:
> It seems weird that 32-bit depth images for the splash screen become corrupted like that, when they had previously worked with the old layout.

It's not the same PNG loader code that is used.

gagarinlg, 2022-09-02:
> The broader question is probably: can we come up with some code mods that would effect proper color mapping?

I think the message from the PNG lib people was something like: less code, less compatibility, less complexity instead of a full blown 100% compatible, but very complex parser.
I never did PNG, so I have no idea how complex it would be to add more compatibility

> But, hey. Physically, I don't see any differences (that are significant enough to care about) between 32-bit or 24-bit on my RadioMaster's display panel, anyways. So, I'm happy to roll with 24-bit.

The display has a 16 bit interface.

PR #2264 (2022-09-05): fix: splash image pixel format
Fixes #2261 

Summary of changes:
- force pixel format to `RGB565` when loading splash image
closed 2022-09-05 by pfeerick