EdgeTX/edgetx #2262: CRSF trainee input

gagarinlg, 2022-09-02:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

As more and more radios have ExpressLRS build in, a wireless trainer setup requires an external module on the trainee radio, when it has an internal ExpressLRS module.

### Describe the solution you'd like

Add CRSF as a trainer input mode for the trainer plug and the external module slot, so that ExpressLRS receivers can be used on the trainer radio to create a wireless trainer setup.

pfeerick, 2022-09-02:
It is probably in need of some independent hardware testing and tweaking (ie. Raphael's comment about not keeping on adding new modes)

https://github.com/EdgeTX/edgetx/pull/1861

raphaelcoeffic, 2022-09-03:
Supporting CRSF Trainer on some serial input should be simple, and yes, we need to add a parameter for the protocol, as we are otherwise having more and more modes.

Regarding the serial input itself, we need a non-inverted UART, which is not available at the back of any radios, especially considering that at +400kbps, we don’t have the option of using software serial.

gagarinlg, 2022-09-03:
Do we need to support high uart speeds?
When you can not fly solo you probably do not get tge difference between 50 Hz and 500 Hz rates

raphaelcoeffic, 2022-09-03:
> Do we need to support high uart speeds?
> When you can not fly solo you probably do not get tge difference between 50 Hz and 500 Hz rates

That is not the point: CRSF serial Rx use 420kbps by default, and I’m not sure you configure them differently.

gagarinlg, 2022-09-03:
Oh, I was not aware of that.