EdgeTX/edgetx #2263: Bind "Beep" never stops when binding Radiolink and SFHSS - carries on chirping - FrSky protocol Rx's are OK.

MirlynG, 2022-09-02:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Brand new TX16s MkII. Flashed with 2.7.1 and MTM 1.3.3.14. 

When binding a Radiolink, or SFHSS Rx, the bind tone continues to chirp. Binding works fine. Chirping does not stop by touching the bind button. You can select another model while it is chirping and it does not stop. If you try to bind another model it does not stop, unless it has a D8/16 Rx when it will continue to chirp until D8/16 Rx has bound. Then it will stop. 
If you select aV761 Rx it will stop when Internal Rf is selected. No need to bind.
The only other way to stop it, is to switch off! On restarting, it has stopped.

I imported everything from MkI TX16s running the same firmware. Did not have the issue on the old Tx. 
Have tried re-flashing both firmwares (Edge and MTM). Also went back to MTM 1.3.3.7. I also flashed today's EdgeTx nightly.
Nothing stopped the problem.

### Expected Behavior

Binding tone expected to stop after when bound, on time out or pressing bind button.

### Steps To Reproduce

Select a Radiolink or SFHSS Rx.
Go through bind process. 
Tone does not switch off.
Select new model, with same Rx, does not stop
Select and bind a D8/16Rx and it will stop when bound.
Select a V761 Rx and it stops when the Internal Rf is selected. Do not need to bind with this.

### Version

2.7.1

### Transmitter

Radiomaster TX16S

shawnholc, 2022-10-03:
will someone fix this please i have the same issue as well. i flew the airplane like this the other day with the bind beep going on i just turned down the volume some.. and i got a few low signal reports and i was less than 100m away. using TX16S MKII EdgeTX and latest 4in1, Radiolink R8FM with Radiolink BYME-A flight controller. beep, beep, beep! beep, beep, beep!

pfeerick, 2022-10-03:
Ok, I can reproduce this at last, as the RM WT01 (wireless trainer) supports Futaba / SFHSS. Binding beep keeps going regardless of bind completion, and doesn't stop if you press the bind button making it inactive.

pfeerick, 2022-10-03:
>  i got a few low signal reports and i was less than 100m away.

Make sure you've done the RF tune for your MPM - both SFHSS and Radiolink use the CC2500 and thus show the RF Freq fine tune option for calibration. 

https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/docs/Frequency_Tuning.md

shawnholc, 2022-10-03:
yea, i did that after i was losing signal, and it really didnt help much. the beep, beep, beep was still there nothing to fix that unless someone fixes the "bug"

pfeerick, 2022-10-03:
That won't fix the beep, but it should fix the range unless the MPM is incorrectly in bind mode or something.

raphaelcoeffic, 2022-10-03:
@pfeerick is that the same on current `main` / `2.8`?

raphaelcoeffic, 2022-10-03:
This happens when the MPM does not detect the end of the binding. Possibly some protocol do not send anything when binding ended properly. It seems the UI code was not resetting the MPM binding status properly, only the general one.

shawnholc, 2022-10-03:
so how do i download the new version? sorry im a noob.

MirlynG, 2022-10-03:
I would not be inclined to download the current release candidate of 2.8 yet. It has way too many issues that need resolving yet! Better to just put up with it until 2.8 is more stable.

shawnholc, 2022-10-03:
i went ahead and flashed it, works good. just need to click bind again once its bound up.. is it possible to get this fix in 2.7.1?
this version is pretty strange..

gagarinlg, 2022-10-03:
No more 2.7 releases. 2.8 will be out soon.

pfeerick, 2022-10-03:
However, the fix is pretty simple, so you could just add it to 2.7.1 yourself and build a custom build if you want that. 

Do you actually need to press the bind button in the first place... I'm not familiar with Futaba SFHSS, but the info and guides I've seen so far say it is an auto-bind protocol - i.e. you turn the transmitter on, and you press the bind button on the RX, and it just binds. No need to press bind on the transmiter side at all.

shawnholc, 2022-10-03:
hmmm... pretty sure i had to press the bind button... let me check.. yea you have to press bind button

shawnholc, 2022-10-03:
anyhow, thank you for fixing it for us.

pfeerick, 2022-10-03:
Thank you to you both for reporting it, and also for testing the fix :)

MirlynG, 2022-10-03:
> However, the fix is pretty simple, so you could just add it to 2.7.1 yourself and build a custom build if you want that.
> 
Thanks for the help guys, as ever but unsure how this works. Do I just copy the relevant tx16s.bin file from the [edgetx-firmware-merge] zip file to my sd card and flash it or is the process more complicated? Does the commando8-20169b3.bin file have to be flashed as well?

gagarinlg, 2022-10-03:
If you have a TX16S, copy the TX16S file to your SD card and flash it.
As always, make a backup of your SD card first, to have a way back.

MirlynG, 2022-10-03:
Great. Thank you so much.

shawnholc, 2022-10-04:
how do you add the fix to the 2.7.1 and make a custom build?

MirlynG, 2022-10-04:
See gagarinig's post above.

MirlynG, 2022-10-04:
> how do you add the fix to the 2.7.1 and make a custom build?

Check out my last post and the response posted by "gagarinlg" above.
Here's the zip file download link: https://github.com/EdgeTX/edgetx/suites/8580596495/artifacts/384199955

shawnholc, 2022-10-05:
i did that buy it put 2.8 on my radio.

MirlynG, 2022-10-05:
> i did that but it put 2.8 on my radio.
Yeah. Did that to me too. Anybody have a way around this?

shawnholc, 2022-10-05:
i used to compile blender 3d several years ago, but i forgot most of it.

pfeerick, 2022-10-06:
I could do it for you, but I would prefer to help you fish for yourself... since most of the work has been done so you *can* build it yourself quite easily ;)

So have a look at this: https://github.com/EdgeTX/edgetx/wiki/Building-radio-firmware-in-a-webbrowser-with-Gitpod

There is a written guide as well as a video. 

You'll have a web-based code editor - you can make changes to the source before you build the files - so make the relevant fixes from https://github.com/EdgeTX/edgetx/pull/2455  - i.e. if you are using TX16, you don't need anything except the colorlcd change. Or the relevant b&w fix if using a b&w radio - there you just need the right screen size. 

Since I'm pretty sure you want 2.7, the catch will be there have been a lot of changes to this part of the code, so I would suggest you want to make the change needed for colorlcd in model_setup.cpp at around line L1088 ;) 

If you want to give it a try and get stuck, yell out ;)

PR #2455 (2022-10-06): fix(multi): set bind status explicitly when aborted manually
Fixes #2263
commit f7e3458b (2022-10-06): fix(multi): set bind status explicitly when aborted manually (#2455)

* fix(multi): set bind status explicitly when aborted manually

Fixes #2263

* fix: adjust MPM bind status on B&W as well
commit 5efadccd (2022-10-08): fix(multi): set bind status explicitly when aborted manually (#2455)

* fix(multi): set bind status explicitly when aborted manually

Fixes #2263

* fix: adjust MPM bind status on B&W as well
closed 2022-10-06 by pfeerick