EdgeTX/edgetx #2267: Long press roller doesn't bring menu on betaflight Lua script

nangel0, 2022-09-03:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

The betaflight Lua script long press menu doesn't appear on roller long  press on my tx16s. 

Latest nightly on tx16s
Latest nightly on Lua script 
3.0.0 RC2 

### Expected Behavior

 be hable to save and reboot from Lua script menu when roller long press 

### Steps To Reproduce

Enter betaflight Lua script 
Adjust pids with betaflight Lua script
Long Press roller does nothing (should appear menu) 

### Version

Nightly (Please give date/commit below)

### Transmitter

Radiomaster TX16S

pfeerick, 2022-09-04:
I can reproduce this - using nightly EdgeTX, and release versions of Betaflight and Betaflight Lua, with ELRS 2.5.1.

nangel0, 2022-09-04:
However my jumper t12 with same versions behaves correctly 👍

commit 96f16320 (2022-09-25): fix(lua): handle ENTER_LONG properly

Fixes #2267
commit c1afe755 (2022-09-25): fix(lua): handle ENTER_LONG properly

Fixes #2267
commit 228f5774 (2022-09-25): fix(lua): handle ENTER_LONG properly

Fixes #2267
PR #2353 (2022-09-25): fix(lua): handle ENTER_LONG properly
Fixes #2267
closed 2022-09-25 by pfeerick