EdgeTX/edgetx #2268: Pot&Slider notices positions are always actives in Companion

luzrael, 2022-09-04:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Companion

### Current Behavior

When reading the radio models and settings, the potentiometers checks are always active, regardless of how they are configured in the radio

### Expected Behavior

The potentiometers checks of a model in Companion should match those of the radio.

### Steps To Reproduce

1. Turn on the radio.
2. Connect the radio to the Companion.
3. Read models and settings from the radio
4. Open any model/template readed.
5. In the first tab, the potentiometers checks are always active, regardless of how they are configured in the radio

### Version

2.7.1

### Transmitter

Radiomaster TX16S

### Anything else?

[image]
[image]
[models.zip](https://github.com/EdgeTX/edgetx/files/9483702/models.zip)

elecpower, 2022-09-04:
Fixed in 2.8

pfeerick, 2022-09-04:
Indeed. Was fixed by https://github.com/EdgeTX/edgetx/pull/1981 so is present in a nightly build, or in 2.8 when released.

closed 2022-09-04 by pfeerick