EdgeTX/edgetx #2272: Mdl and Sys buttons Strange behavior

KirkTz, 2022-09-04:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Hi until today my Tx16s worked  flawless with the 2.71 ,now i have a strange behavior  from the buttons Mdl and Sys
When i press them the radio doesn't open the menus for model and system settings ,i have to press the screen and choose the model settings from there ,sometimes works as is should ,but in the majority no 
I have change the sd card for a new 32Gb with no luck
Any suggestions ? 
Thnx 

### Expected Behavior

 Mdl and Sys opening menus

### Steps To Reproduce

same problem as many times reboot radio

### Version

2.7.1

### Transmitter

Radiomaster TX16S

pfeerick, 2022-09-05:
SD card (and contents) should have generally not effect on this as it's only used for settings - not for general operation of the radio. This is sounding more like a hardware issue if it suddenly started happening and you've not recently changed the firmware. Does `Radio Settings -> Hardware (second last tab) -> Keys (at the bottom)` detect and MDL and SYS keys being pressed (i.e. they should be changing from `0` (not pressed) to `1` (pressed). If not, it really sounds like a hardware issue. Might be worth double-checking the other hardkeys while you are there (the ones in the left column) while you're there.

KirkTz, 2022-09-05:
The buttons are working for sure because after 4-5sec  after pressing the mdl  the radio displays the models photos for selecting a different model (sorry i forgot to mention that)  i will check your suggestion later tonight

pfeerick, 2022-09-05:
If you can catch it playing up again, see if they show up in that `hardware => keys` page. If you're getting the MDL select screen with MDL, it sounds like you're holding the button down too long, or the button is getting stuck. ;) the SYS key doesn't have a second function yet, so both long and short press are the same for it.

KirkTz, 2022-09-06:
> If you can catch it playing up again, see if they show up in that `hardware => keys` page. If you're getting the MDL select screen with MDL, it sounds like you're holding the button down too long, [...]

ok i will check it thank you

closed 2023-05-26 by gagarinlg