EdgeTX/edgetx #2278: Widget settings not saved

DavBfr, 2022-09-05:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Changing a widget setting and powering off the radio won't update the model YAML file

### Expected Behavior

Widget settings are scheduled to be saved whenever the widget settings dialog is dismissed.

### Steps To Reproduce

1. Open the settings for a widget
2. Change a setting
3. Close the dialog
4. Wait a few seconds
5. power off the radio

### Version

Nightly (Please give date/commit below)

### Transmitter

Radiomaster TX16S

### Anything else?

Commit b4e1ceb4f

gagarinlg, 2023-09-18:
This should be fiyed by now, if I remember correctly. Can anyone test with 2.9

pfeerick, 2023-09-24:
AFAICT, this is gone now (I do vaguely remember a PR targeting this issue)... I changed settings for the Outputs and Values widgets (one at a time), closed the dialog, powered off the radio (didn't even wait) and settings were correct on powering back on.

closed 2023-09-24 by pfeerick