EdgeTX/edgetx #2284: Special Functions Nighty 2.8

Mikey1968, 2022-09-06:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

This is in Special Functions
When I'm in there clicking on the little folder icon there's a delay opening the folder for the sounds folder. Also you can't scroll backwards to go Z to A like in previous versions

### Expected Behavior

Maybe getting fixed

### Steps To Reproduce

Not sure what you mean 

### Version

Nightly (Please give date/commit below)

### Transmitter

Radiomaster TX16S

banyaszg, 2022-09-27:
The backward scroll part seems to be a general list handling difference in 2.8.

banyaszg, 2022-09-27:
@pfeerick Is it intentional, or should I fix it?

pfeerick, 2022-09-27:
> The backward scroll part seems to be a general list handling difference in 2.8.

Correct. As I commented to Mikey when raised another time, I'm not sure there is much need for that since you can get to the other end of the list pretty quickly on pretty much all (if not all) colorlcd radios (i.e. either via touch, or via rotary encoder). One disadvantage of it basically being an endless list would be if you scroll fast, it'll just wrap around instead of stopping at the end. So I'm inclined to leave it as it is, unless it's something a majority of users want.

banyaszg, 2022-09-27:
I see your point and agree.

pfeerick, 2022-10-10:
I'm going to assume/suggest `FileChoice::openMenu()` is slower than it could be (or at least than it was). However, the code seems ok though since the file processing and sorting are done separately from the menu being constructed. If it can't be sped up again, then probably need some loading indicator shown so it's understood the radio is responding to the request.

pfeerick, 2022-10-10:
With the same radio, same SD card, same standard En sound pack:

2.7.1 => https://user-images.githubusercontent.com/5500713/194844873-0ed74f70-5eac-4f38-83d8-eae4d54bb005.mp4

2.8.0-RC2 => https://user-images.githubusercontent.com/5500713/194845016-4757ddd1-a4dd-4919-a6a4-5aa4c6bb5e52.mp4

gagarinlg, 2022-10-10:
My assumption is, that LVGL recalculates the layout every time you add a new element. So it is doing a lot of unnecessary work. I tried multiple times to find a way to disable that and then let it do the layout update once in the end, but I could not find a way to do this.

banyaszg, 2022-10-10:
What about setting the table row count first, and add the line values after that? That way it won't auto expand.

gagarinlg, 2022-10-10:
The row height is dynamic. So this would (probably) not help much, but I not try it.

banyaszg, 2022-10-10:
There is an empty line added to the beginning and it has the same size. So I assume the content doesn't count, just the cell style, padding, and the cell numbers.

banyaszg, 2022-10-10:
I implemented a test to try these, just I don't have my radio with me. So I'll do later at home and let you know about the result.

banyaszg, 2022-10-10:
I tested my modifications, and it seems faster. I think the waiting time is halved.
I'll create a draft PR after the family todos.

Eldenroot, 2022-10-10:
And what about this part of the ticket?

`Also you can't scroll backwards to go Z to A like in previous versions`

PR #2516 (2022-10-12): Speed up bulk menu addition.
Fixes #2284

Summary of changes:
Speed up menu initializítion by delaying the UI update and do it in one step at the end.
Modified parts:
- [x] ModelSetupPage::build()
- [x] ModelsPageBody::editLabels()
- [x] ModelCurvesPage::build()
- [x] ModelMixesPage::newMix()

Requires [libopenui#77](https://github.com/EdgeTX/libopenui/pull/77)
closed 2022-10-12 by pfeerick